verilator/verilator #3522: Internal develop-v5 clang-tidy warnings

wsnyder, 2022-07-30:
@gezalore you might want to look at 12925cd8b where I pushed some clang-tidy fixes on new code in develop-v5.

In addition you should look at these.

```
940f946f67 (Wilson Snyder  2022-07-30 12:29:54 -0400 1526)                                             externalDomains);
src/V3Order.cpp:1526:45: warning: parameter 'externalDomains' is passed by value and only copied once; consider moving it to avoid unnecessary copies [performance-unnecessary-value-param]

940f946f67 (Wilson Snyder  2022-07-30 12:29:54 -0400  954)                  const string& tag, bool slow, V3Order::ExternalDomainsProvider externalDomains)
src/V3Order.cpp:954:18: warning: pass by value and use std::move [modernize-pass-by-value]

940f946f67 (Wilson Snyder  2022-07-30 12:29:54 -0400  954)                  const string& tag, bool slow, V3Order::ExternalDomainsProvider externalDomains)
src/V3Order.cpp:954:48: warning: pass by value and use std::move [modernize-pass-by-value]

940f946f67 (Wilson Snyder  2022-07-30 12:29:54 -0400  957)         , m_externalDomains{externalDomains}
src/V3Order.cpp:957:29: warning: parameter 'externalDomains' is passed by value and only copied once; consider moving it to avoid unnecessary copies [performance-unnecessary-value-param]

940f946f67 (Wilson Snyder  2022-07-30 12:29:54 -0400  974)         OrderProcess visitor{netlistp, graph, trigToSen, tag, slow, externalDomains};
src/V3Order.cpp:974:69: warning: parameter 'externalDomains' is passed by value and only copied once; consider moving it to avoid unnecessary copies [performance-unnecessary-value-param]

```

Make clang-tidy runs this, it's hacky and we're intentionally far from clean so no requirement this be clean.

gezalore, 2022-08-02:
Fixed, though they are not really an issue (a handful of extra copies).

BTW I assumed that was clang-tidy-11. V3Order.cpp is clean (with or without the fix) using clang-tidy-14 (Ubuntu 22.04 default).

closed 2022-08-02 by gezalore

gezalore, 2022-08-02:
> V3Order.cpp is clean (with or without the fix) using clang-tidy-14 (Ubuntu 22.04 default).

That looks a bit suspicions actually, as most all files are clean with clang-tidy-14.