verilator/verilator #3525: Question regarding constifyCpp pass.

kamilrakoczy, 2022-08-02:
I've noticed that executing constifyCpp pass multiple times in a row results in different output files.
I wounder, is this desired behavior to reduce memory/time usage or it is some kind of a bug that doesn't take into account newly created nodes?

Whole patch that I'm using when comparing output between master revision:
```
diff --git a/src/Verilator.cpp b/src/Verilator.cpp
index f28faabf6..750f6ccf2 100644
--- a/src/Verilator.cpp
+++ b/src/Verilator.cpp
@@ -460,6 +460,7 @@ static void process() {
     if (!v3Global.opt.xmlOnly() && v3Global.opt.fSubst()) {
         // Constant folding of expanded stuff
         V3Const::constifyCpp(v3Global.rootp());
+        V3Const::constifyCpp(v3Global.rootp());
         V3Subst::substituteAll(v3Global.rootp());
     }
 
```

gezalore, 2022-08-02:
I assume the difference is not big between running it once or twice?

It would be ideal for V3Const to always run to fixed point, unless this is prohibitively expensive (I don't think it should be for most transformations there). There definitely should be a fixed point (or it's a bug). A patch would be welcome. 

A lot of rules are either:
1. 'iterateChildren', then optimize this node (i.e.: bottom up), or
2. change this node, then 'iterate' the new node (i.e.: top-down).

Some rules differ (possibly the author just forgot the iteration calls), and also some look at the `backp()`, so even the bottom up rules might not be always conclusive.

wsnyder, 2022-08-03:
Currently for performance reasons it is intentional that not all optimizations may be performed in one pass, as some optimizations require looking down around 4 children deep, so to do otherwise would require reiterating on the great-grandparent which is potentially expensive.  The assumption is that the next time through V3Const between some later stages will likely get the optimization.  It certainly is possible this could be cleaned up/revisited but hasn't been a priority.

closed 2022-10-19 by wsnyder