verilator/verilator #3526: Is there a way to convert V3Graph back to HDL or C++ code?

burntfalafel, 2022-08-04:
Currently the V3Trace block creates V3Graph edges, and optimises the graph. I was wondering is there any to make the graph spit back code? Even the generated C++ model will do.

All I can emit from the V3Graoh block is the `*.dot` and `*.tree` files...

Thanks!

wsnyder, 2022-08-04:
Not sure what you mean by code, what are you trying to do?

burntfalafel, 2022-08-04:
The aim was to see the code output from the graph optimisation passes done in the V3Trace module. (here:) https://github.com/verilator/verilator/blob/f9f66d787e5b386c89cb9c9714bcf18a55c4090d/src/V3Trace.cpp#L729

**Reason?** Was trying to get the optimised HDL code (or optimised C++ code). 

At the moment I can only dump out the optimised graph in `*.dot` which only allows me to visualise the optmisation difference.

burntfalafel, 2022-08-06:
Apologies, this question was vague. Got confused by the order of execution of cpp files in the project. Solved by increasing verbosity.

closed 2022-08-06 by burntfalafel