verilator/verilator #3528: VCD only contains signal definitions when building and running examples/cmake_tracing_sc

k0nze, 2022-08-04:
Hi,

I'm having issues getting signal data in VCDs when using SystemC 2.3.3, Verilator 4.224, and building with CMake. I compile/install SystemC and Verilator with the following commands on Ubuntu 20.04:

```
# compile/install systemc
mkdir systemc-2.3.3
export SYSTEMC_HOME=$(pwd)/systemc-2.3.3
git clone https://github.com/accellera-official/systemc
cd systemc
git checkout 2.3.3
mkdir build
cd build
cmake -DENABLE_PTHREADS=ON -DCMAKE_CXX_STANDARD=17 ..
make -j4
make install
cd ../..

# compile verilator
export VERILATOR_ROOT=$(pwd)/verilator
git clone https://github.com/verilator/verilator
cd verilator
git checkout v4.224
autoconf
./configure
make -j4
```

When compiling and running the `example/cmake_tracing_sc` with the following commands:
```
cd examples/cmake_tracing_sc
mkdir build
cd build
cmake ..
make -j4
./example +trace
```

the `logs/vlt_dump.vcd` only contains the signal definitions:
```
$version Generated by VerilatedVcd $end
$date Thu Aug  4 15:09:53 2022 $end
$timescale 1ps $end

 $scope module top $end
  $scope module top $end
   $var wire  1 # clk $end
   $var wire  1 $ fastclk $end
   $var wire 40 - in_quad [39:0] $end
   $var wire  2 , in_small [1:0] $end
   $var wire 70 / in_wide [69:0] $end
   $var wire 40 ' out_quad [39:0] $end
   $var wire  2 & out_small [1:0] $end
   $var wire 70 ) out_wide [69:0] $end
   $var wire  1 % reset_l $end
   $scope module sub $end
    $var wire  1 # clk $end
    $var wire 32 3 count_c [31:0] $end
    $var wire 32 2 count_f [31:0] $end
    $var wire  1 $ fastclk $end
    $var wire  1 % reset_l $end
   $upscope $end
  $upscope $end
 $upscope $end
$enddefinitions $end
```

GTKWave complains when I try to open `logs/vlt_dump.vcd`:
```
[-1] start time.
[-1] end time.
VCD times range is equal to zero.  Exiting.
GTKWAVE | File load failure, new tab not created.
```

Do I miss something during the compilation and installation of SystemC or Verilator? 
Thanks in advance.

yTakatsukasa, 2022-08-04:
Can you try building SystemC with adding `-DENABLE_PHASE_CALLBACKS_TRACING=OFF` cmake as below?

> cmake -DENABLE_PTHREADS=ON -DCMAKE_CXX_STANDARD=17 -DENABLE_PHASE_CALLBACKS_TRACING=OFF ..

See also this [faq](https://github.com/verilator/verilator/blob/master/docs/guide/faq.rst#how-do-i-generate-waveforms-traces-in-systemc)

k0nze, 2022-08-05:
Perfect now it works!

closed 2022-08-05 by k0nze