verilator/verilator #3533: char * will become c++m symbol

quantrpeter, 2022-08-07:
Hi
   If using "char *" in dpi function, the function will become c++ symbol, rather than c

```
import "DPI-C" function int add(input int a, input int b);
import "DPI-C" function void profile (input string str);
```

```
int add(int a, int b)
{
	printf("add ar\n");
	return a + b;
}

void profile(char *str){
	printf("quantr %s\n", str);
}
```

nm ./obj_dir/sim_main.o|grep 'add\|profile'

```
0000000000000028 T add
0000000000000042 T _Z7profilePc
```

Then if i link it, it will said:

/usr/bin/ld: Vquantr_i_main__ALL.a(Vquantr_i_main___024unit__DepSet_h947cdc23__0.o): in function `Vquantr_i_main___024unit____Vdpiimwrap_profile_TOP____024unit(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
Vquantr_i_main___024unit__DepSet_h947cdc23__0.cpp:(.text+0x11): undefined reference to `profile'

thanks

wsnyder, 2022-08-07:
You need

    extern "C" {

around your C function.  This is unrelated to Verilator, true for all simulators.

closed 2022-08-07 by wsnyder

quantrpeter, 2022-08-07:
nono, i solved it by changing

``
void profile(char *str){
	printf("quantr %s\n", str);
}
```

to
```
void profile(const char *str){
	printf("quantr %s\n", str);
}
```

it need "const"