verilator/verilator #3534: dpi 64 bits

quantrpeter, 2022-08-07:
Hi, DPI seems don't have long (64 bits), is it official way to use chandle to represent 64 bits? 

```
import "DPI-C" function void profile (input string str, input chandle value);
```

how to suppress this warning? I want to write a DPI function can accept 1-64 bits value

```
%Warning-WIDTH: src-output/id.v:1318:1: Operator TASKREF 'profile' expects 64 bits on the Function Argument, but Function Argument's VARREF 'wd_o' generates 5 bits.
```

thanks
Peter

gezalore, 2022-08-07:
It's called `longint`, but this is just basic SystemVerilog syntax. Please refer to the SystemVerilog standard, there are also example in the test suite, for example; 
https://github.com/verilator/verilator/blob/master/test_regress/t/t_dpi_arg_input_type.v

closed 2022-08-07 by gezalore

quantrpeter, 2022-08-07:
thank you my hero