verilator/verilator #3536: Segfault on invalid port declaration

1425, 2022-08-09:
Running "verilator --cc 2.v" where 2.v contains "module d(0var[0:0]e=0" will produce a segfault.  

This has been tested with:

Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
Verilator 4.225 devel rev v4.224-82-gcbe1b8e26

On Ubuntu 22.04.1 LTS.

This seems to be fixed adding a null check on line 2025 of verilog.y:
-                        { $$ = VARDONEA($<fl>1,*$1,$2,$3); $$->valuep($5); }
+                        { $$ = VARDONEA($<fl>1,*$1,$2,$3); if($$) $$->valuep($5); }

wsnyder, 2022-08-09:
Sorry, but I am not personally interested in spending time on bugs from fuzzing, provided there is at least an error emitted before the segfault.  The "fuzzer" directory in the kit contains "actual_fail" which can be used to find "interesting" failures, that is those items that segfault without error.

closed 2022-08-09 by wsnyder

1425, 2022-08-09:
Would that calculus be different if I sent a pull request?

wsnyder, 2022-08-09:
If you want to make complete patch sets, that is debug them and also add tests so all the cases are covered, I'll review and merge once appropriate.

When I ran a rough check years ago there was more than 100 of these to fix, so know what you're signing up for ;)