verilator/verilator #3537: Increase Maximum Number of Warnings of Verilator

semichipper, 2022-08-09:
We are regular Verilator users. Among other things, we use it for LINTing purposes.

Verilator has default 50 warnings after which it stops processing and errors out, citing that it has already processed enough number of warnings.

Is there a way or a switch through which we can change this - so we can let Verilator throw more warnings before it stops processing...

(We have a case where we have many WIDTH warnings that are totally harmless. We want to go after more things down the line.).

Regards,

gezalore, 2022-08-09:
Although you talk about warnings, you are likely looking for the `--error-limit` option: https://verilator.org/guide/latest/exe_verilator.html?r%20limit#cmdoption-error-limit
Feel free to re-open and follow up if that does not work for you.

closed 2022-08-09 by gezalore

wsnyder, 2022-08-09:
Also I would suggest using -Wno-WIDTH as an easier way of getting past those specific warnings.

semichipper, 2022-08-09:
Thanks @gezalore and @wsnyder . Will try them and come back if required.
Yes, we could waive them using -Wno-WIDTH. Just didn't want to put into the flow just yet.

Regards.