verilator/verilator #3538: Command line options for finding `systemc_header include files

jjones-github, 2022-08-09:
I have a rather basic setup and I suspect I am missing some sort of argument(s) on the command line to get this to work properly. This is somewhat similar to the test_regress/t/t_extend_class.v and t_extend_class_c.h but I had trouble figuring you the command line options. Any and all help is greatly appreciated.

This is what I use to run `verilator -Wno-fatal --cc --exe --top-module t -sv t.sv s--build sim_t.cpp && ./obj_dir/Vt`

The error I see is 

_In file included from Vt__ALL.cpp:3:_
_In file included from ./Vt.cpp:5:_
_In file included from ./Vt__Syms.h:18:_
_./Vt_t.h:15:10: fatal error: 't.cpp' file not found_
_#include "t.cpp"_
_         ^~~~~~~_
_1 error generated._
_make: *** [Vt__ALL.o] Error 1_

I had trouble escaping backticks so I inserted three files all together; t.cpp, sim_t.cpp, and then t.sv. I placed a comment in the code section below to differentiate each file. 

`````````````
// t.cpp
class Accum {
public:
    Accum() {}
    ~Accum() {}

    vluint32_t updateAccum(vluint32_t a, vluint8_t c) {
        return a + c;
    }

    void updateAccumByRef(vluint32_t& a, vluint8_t c) {
        a = a + c;
    }

};

// sim_t.cpp

//#define TRACES
#include <iostream>
#include <memory>
#ifdef TRACES
#include "verilated_fst_c.h"
#endif
#include <verilated.h>
#include "Vt.h"
#define COUNTS_TO_RUN 15

double sc_time_stamp() { return 0; }
int main(int argc, char** argv, char** env) {
    if (false && argc && argv && env) {}
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    contextp->debug(0);
    contextp->randReset(2);
#ifdef TRACES
    contextp->traceEverOn(true);
    VerilatedFstC* tfp = new VerilatedFstC;
#endif
    contextp->commandArgs(argc, argv);
    const std::unique_ptr<Vt> top{new Vt{contextp.get(), "TOP"}};
    top->rstN = !0;
    top->clk = 1;
    top->counterInVld = 0;
    while(contextp->time() < COUNTS_TO_RUN) {
        contextp->timeInc(1);
        top->clk = !top->clk;

        if (!top->clk && contextp->time() < 12) {
            if (contextp->time() > 1 && contextp->time() < 10) {
                top->rstN = !1;  // Assert reset
            } else {
                top->rstN = !0;  // Deassert reset
            }
        }
        if (!top->clk && top->rstN && contextp->time() >= 10) {
            top->counterInData = 1;
            top->counterInVld = 1;
        }
        if (top->clk && top->rstN) {
            std::cout << "output = " << top->accumulator << std::endl;
        }

        top->eval();
    }
#ifdef TRACES
    tfp->flush();
#endif
    top->final();
    return 0;
}

// t.sv
`timescale 1ns/1ns

module t (
    input counterInVld,
    input [7:0] counterInData,

    output logic [29:0] accumulator,

    input clk, rstN
);

`ifdef _ENABLE_TRACES_
initial begin
    $dumpfile("sim.fst");
    $dumpvars();
end
`endif

`ifdef _USE_CPP_
`systemc_header
#include "t.cpp"
`systemc_interface
Accum* m_myobjp;
`systemc_ctor
m_myobjp = new Accum();
`systemc_dtor
delete m_myobjp;
`verilog
`endif

logic [29:0] counter;

always_ff @(posedge clk or negedge rstN) begin
    if(~rstN) begin
        counter <= '0;
    end else if (counterInVld) begin
`ifdef _USE_CPP_
        counter <= $c("this->m_myobjp->updateAccum(",counter, counterInData,")");
`endif
        counter <= counter + { {($bits(accumulator)-$bits(counterInData)){1'b0}}, counterInData};
    end
end

always_comb begin
    accumulator = counter;
end

endmodule
`````````````

gezalore, 2022-08-09:
That is an error from the C++ compiler invoked by `--build`. Try passing `-CFLAGS "-I <abs-path-to-where-t.cpp-lives>"` on your verilator command line.

https://verilator.org/guide/latest/exe_verilator.html?highlight=cflags#cmdoption-CFLAGS

closed 2022-08-09 by jjones-github

gezalore, 2022-08-09:
> Thank you. That helped. I assume the absolute path is required because the build doesn't happen in your current working directory?

That is the reason. See also the `--Mdir` option which I *think* will be your working directory, though I don't use `--build` myself.

jjones-github, 2022-08-09:
> > Thank you. That helped. I assume the absolute path is required because the build doesn't happen in your current working directory?
> 
> That is the reason. See also the `--Mdir` option which I _think_ will be your working directory, though I don't use `--build` myself.

I'll take a look at `--Mdir` as well. I went with a ``--CFLAGS "-I `pwd`"`` for now. And it appears in my zsh `-CFLAGS "-I $(pwd)"` also works.