verilator/verilator #3539: lib-create creates wrapper with non-matching ports (ignores LSB index)

danbone, 2022-08-10:
It appears that verilator is ignoring port LSB index when creating the SV wrapper file for the DPI library.

My sv file test.sv:

`module test (
   input logic [7:4] a,
   output logic [3:0] b
   );

   assign b = ~a;

endmodule
`

Command:
verilator --lib-create testlib -cc -CFLAGS '-std=c++14' test.sv

Entity in output wrapper file under obj_dir/testlib.sv
`module testlib (
        input logic [7:0]  a
        , output logic [3:0]  b
    );
    
`

You can see that port A is now 8 bits wide

Verilator verision:
Verilator 4.221 devel rev v4.220-26-g880a9be

Using CentOS-6
#1 SMP Wed Aug 7 18:08:02 UTC 2019 x86_64 x86_64 GNU/Linux

wsnyder, 2022-08-10:
The storage in C++ after Verilation always has bit zero as the LSB, so it might be instead that the MSB should be changed to 4.

Might you be able to provide a pull request that updates the test cases to check this, and provide a fix?

danbone, 2022-08-10:
I can certainly try

closed 2022-10-18 by wsnyder