verilator/verilator #3540: Compiling verilator models with Visual STudio

pozzonim, 2022-08-11:
WIth the last verilator version,if I compile the created .cpp files  with Visual STudio 2017 I get the error;

Verilator requires a C++11 or newer compiler

wsnyder, 2022-08-11:
Please look at the code that prints that message in verilatedos.h, you'll need to determine what macros VS2017 is setting to fix the message, presuming it is 2011 compatible, then if it works please consider submitting a pull request.

pozzonim, 2022-08-11:
The issue is caused by this code in verilatedos.h

#if __cplusplus >= 201103L || defined(__GXX_EXPERIMENTAL_CXX0X__) || defined(VL_CPPCHECK)
# ifndef VL_NO_LEGACY
// These are deprecated historical defines. We leave them in case users referenced them.
#  define VL_EQ_DELETE = delete
#  define vl_unique_ptr std::unique_ptr
#  define vl_unordered_map std::unordered_map
#  define vl_unordered_set std::unordered_set
#  define VL_INCLUDE_UNORDERED_MAP <unordered_map>
#  define VL_INCLUDE_UNORDERED_SET <unordered_set>
#  define VL_FINAL final
#  define VL_MUTABLE mutable
#  define VL_OVERRIDE override
# endif
#else
# error "Verilator requires a C++11 or newer compiler"
#endif

wsnyder, 2022-08-11:
Yes, the point is you need to figure out the value of the defines for your compiler, then submit a fix.

pozzonim, 2022-08-11:
What trigger the error is the following:
#if __cplusplus >= 201103L || defined(__GXX_EXPERIMENTAL_CXX0X__) || defined(VL_CPPCHECK)
which is not satisfied in my Visual Studio environment.

Said that, I don't have enough knowledge to understand what I can do to fix the issue.

gezalore, 2022-08-11:
Your problem is that `__cplusplus >= 201103` is false, which is a standard define available in C++11, so it seems you are compiling without C++11 support, or the compiler is otherwise broken.

pozzonim, 2022-08-11:
Thanks,
this comment allowed me to solve the issue:

I added the following option to the Visual STudio compiler : /Zc:__cplusplus 

This solved the issue.

Otherwise, by default, Visual studio returns 199711L as __cplusplus and the check fails.

gezalore, 2022-08-11:
Needing that seems like a sad trait of msvc, but glad you figured it out. Probably should add that to the docs

closed 2022-08-11 by gezalore