verilator/verilator #3543: Program received signal SIGSEGV, Segmentation fault.

openpowerwtf, 2022-08-14:
I am getting this fail with verilator version v4.224-82-gcbe1b8e26 on Linux (Linux GatorCountry 5.13.0-52-generic #59~20.04.1-Ubuntu SMP Thu Jun 16 21:21:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux).  

Background: I have not seen the SEGV with previous versions of verilator;  I tried several versions back to 4.106 this past week since I have been debugging bad sim results with verilator (and iverilog passing), related to incorrect clocking behavior.  After splitting [clk, rst] into separate signals, I ran ~15000 core cycles successfully today (v4.224-26-g8b7480806).

* repo = https://github.com/OpenPOWERFoundation/a2o

```
cd dev/sim/verilator
verilator -cc  --exe --trace --Mdir obj_dir --language 1364-2001 -Wno-fatal -Wno-LITENDIAN --error-limit 1  -Iverilog/a2o_litex -Iverilog/work -Iverilog/trilib -Iverilog/unisims a2owb.v tb_litex.cpp  |& tee verilator.txt

...

- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_044_life.tree
- V3Table.cpp:431:    tableAll: 
- V3Table.cpp:239:     Table Optimize spacetime=6.4 ALWAYS 0x55575b3a3080 <e88880570> {by572ah}
- V3Simulate.h:367:   Unknown node type in SimulateVisitor: WHILE
- V3Simulate.h:367:   Unknown node type in SimulateVisitor: WHILE
- V3Simulate.h:367:   Unknown node type in SimulateVisitor: WHILE
- V3Simulate.h:367:   Unknown node type in SimulateVisitor: WHILE
- V3Table.cpp:239:     Table Optimize spacetime=5.93623 ALWAYS 0x555708b0a4d0 <e86114748> {lf389ac}
- V3Table.cpp:239:     Table Optimize spacetime=1.6254 ALWAYS 0x555708de2e70 <e86123761> {lf389ac}
- V3Table.cpp:239:     Table Optimize spacetime=1.6254 ALWAYS 0x555708f57810 <e86127594> {lf389ac}
- V3Table.cpp:239:     Table Optimize spacetime=3.10303 ALWAYS 0x5557090c4d10 <e86131111> {lf389ac}
- V3Table.cpp:239:     Table Optimize spacetime=5.93623 ALWAYS 0x555709f4c6e0 <e86177528> {lf389ac}
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_045_table.tree
- V3Const.cpp:3704:   constifyAll: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_046_const.tree
- V3Dead.cpp:468:     deadifyDTypesScoped: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_047_deadDtypesScoped.tree
- V3Active.cpp:609:   activeAll: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_048_active.tree
- V3Split.cpp:1013:   splitAlwaysAll: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_049_split.tree
- V3SplitAs.cpp:197:  splitAsAll: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_050_splitas.tree
- V3TraceDecl.cpp:530:traceDeclAll: 
- V3Ast.cpp:1154:     Dumping obj_dir/Va2owb_051_tracedecl.tree
- V3Force.cpp:305:    forceAll: 
- V3Gate.cpp:1528:    gateAll: 
dot -Tpdf -o ~/a.pdf obj_dir/Va2owb_052_gate_simp.dot

Program received signal SIGSEGV, Segmentation fault.
```

gezalore, 2022-08-15:
I have been around V3Gate recently so I will take a look. Thanks for the bug report.

If you are having issues with clocking, you could consider trying the develop-v5 branch that should be able to handle arbitrary code without modification.

openpowerwtf, 2022-08-15:
Thanks - I know why it was done but it was unusual enough to just get rid of.  I can try that branch with some versions of the original code - it should be a good test.

gezalore, 2022-09-02:
Fun bug:
Instantiation `iuq_ic_ierat  iuq_ic_ierat0`, specifically ports at:
https://github.com/OpenPOWERFoundation/a2o/blob/c309e9145c8c355c8ec5dd0db030e76680331ce5/rel/src/verilog/work/iuq_ic.v#L564-L565
`tsiv` is driven at the bottom:
https://github.com/OpenPOWERFoundation/a2o/blob/c309e9145c8c355c8ec5dd0db030e76680331ce5/rel/src/verilog/work/iuq_ic.v#L1276
so `iuq_ic_ierat0.time_scan_out` drives `iuq_ic_ierat0.time_scan_in`.

Module `iuq_ic_ierat` port `time_scan_out` is driven from instantiation `tri_cam_16x143_1r1w1c  ierat_cam`, specifically ports at:
https://github.com/OpenPOWERFoundation/a2o/blob/c309e9145c8c355c8ec5dd0db030e76680331ce5/rel/src/verilog/work/iuq_ic_ierat.v#L4497-L4498
and in module `tri_cam_16x143_1r1w1c` we have:
https://github.com/OpenPOWERFoundation/a2o/blob/c309e9145c8c355c8ec5dd0db030e76680331ce5/rel/src/verilog/trilib/tri_cam_16x143_1r1w1c.v#L2696

In the end that is `assign a = b` and `assign b = a` which with my last patch to V3Gate, we try to recursively inline the right hand sides and stack overflow.

I will sort out a fix shortly, but you probably didn't mean to do that?

gezalore, 2022-09-02:
I have pushed a fix. You have a few more combinational loops reported by UNOPTFLAT, unfortunately that is reported after the crashing pass.

@wsnyder this can be triggered by some fairly simple circular logic (see test in fix), and sadly it made it into the last release, so is likely to come up again :/

closed 2022-09-02 by gezalore

openpowerwtf, 2022-09-02:
Thanks!  I will take a look and get it working with latest Verilator.  This core was designed years ago for ASIC technologies and could also generate sim/FPGA logic for arrays, etc. (based on a parameter).  The 'trilib' components did the behavioral vs. tech generation; that is why there are lots of test/scan/etc. signals tied or unused in this RTL.   I think the scan outs were usually tied and the scan ins were sent to an attributed signal to avoid sourceless/sinkless net warnings from the compiler.  Sounds like there is at least one bad connection.

commit d42a2d64 (2022-09-02): Fix V3Gate crash on circular logic

The recent patch to defer substitutions on V3Gate crashes on circular
logic that has cycle length >= 3 with all inlineable signals (cycle
length 2 is detected correctly and is not inlined). Fix by stopping
recursion at the loop-back edge.

Fixes #3543