verilator/verilator #3546: Public access to the m_wroteBytes variable in the verilated_vcd_c.h file.

GregoryCarver, 2022-08-16:
I'm trying to specify a maximum size for the vcd file being created, and then close it out when that size has been reached. Maybe this functionality exists already, and I'm unaware of it. If not, it would be simple to write a function that could do this if I had access to the m_wroteBytes variable.  Thanks in advance!

wsnyder, 2022-08-16:
Please see the existing VerilatedVcd::rolloverMB().

closed 2022-08-16 by wsnyder

GregoryCarver, 2022-08-16:
> Please see the existing VerilatedVcd::rolloverMB().

So this function sets a size, and after that size has been reached when creating a vcd, it continues on to another file with an incremented name. Can I use this to stop producing files after the first file has been completed? I'd like to create one file of a certain size, and when that file has been completed, close the vcd.

wsnyder, 2022-08-16:
rolloverMB does rollover as you describe, it does not close the file.  Please feel free to submit a pull request to add add an e.g. closeMB().

GregoryCarver, 2022-08-16:
My apologies, as I understand it once the threshold that is determined by setting m_rolloverMB that is set by rolloverMB() is met , it closes the data file currently being written to and opens the next file with an incremented name. I'd like to stop there after the first file is completed, and not rollover. I'd like to specify a maximum size, and if that size is reached, close the vcd without creating more files. This could be built into rolloverMB or standalone. I'd imagine it would be more useful to build it into rolloverMB. You could specify a size that you want each file to be, m_rollover, but also either a maximum size or file count. Once the combined file size is reached, or file count is reached, close the vcd and stop dumping.

GregoryCarver, 2022-08-17:
So, I've been looking at the code a while and I think I understand now. I thought when using rollover it would automatically create each file until done. It seems you have to call openNext() to trigger the creation of the next file, though. If you don't call that function when using rollover, then only a maximum of two files are created. The first with the header, and then the second openNextImp() call that happens in open(). If I'm understanding correctly. So, if I wanted a single file of a certain size, I could just check against a bool before entering the second openNextImp() call in open.

wsnyder, 2022-08-18:
> I thought when using rollover it would automatically create each file until done. It seems you have to call openNext() to trigger the creation of the next file, though

That certainly was not the intent, the code is also called through dump() which is called every cycle.  There was a test for this setting but it looks like it was lost in one of the tracing rewrites, and should resurrected.

GregoryCarver, 2022-08-18:
I tried it at work and it rolls over as intended I believe, I think I might have just misunderstood it. We are a couple versions behind also, though. Would you be against a getter for m_wroteBytes?

wsnyder, 2022-08-18:
I think Geza's point is similar to mine above, it would be better to add the functionality for Verilator to be given a file size and have it automatically close the file at that size, similar to how rolloverMB does rollover for you at a size, rather than make you do that implementation in the upper level code.

GregoryCarver, 2022-08-18:
@wsnyder Ohhh I think I understand now. So have an internal variable like m_maxFileSize, and set it using closeMB(), similar to how rolloverMB() sets m_rolloverMB. If it isn't 0, the default, then close when that size is reached. If it is 0 then ignore it and allow the vcd size to be unlimited. Am I understanding that right?

wsnyder, 2022-08-18:
Exactly, and related tests.   It probably would be good to add +verilator+trace+close+mb or some similar flag for runtime control, then no C++ hacks are needed.   Extra credit to do that for the rollover MB flag also as a separate pull.  Actually that said I'll file a bug so it isn't lost.