verilator/verilator #3547: String literal in function argument converted to unsigned long long instead of std::string

Grulfen, 2022-08-16:
Found on verilator master branch

A string literal in a function argument is incorrectly converted to `unsigned long long` instead of `std::string`

I needed to put the function in a package and a class, I think to avoid optimazations that hid this problem.

The following verilog code:

`file: tjo_p.sv`
```systemverilog
package tjo_p;
    class Tjo;
        function automatic string broken_str_arg(input string some_string);
            $display(some_string);
            return some_string;
        endfunction
    endclass
endpackage
```
`file: tjo.sv`
```systemverilog
module tjo();

initial begin
    tjo_p::Tjo tc;
    string bar = "baar";
    $display(tc.broken_str_arg("foo"));
    $display(tc.broken_str_arg(bar));
end
endmodule
```

Gets converted into this cpp code:
```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Init
    std::string __Vtask_broken_str_arg__0__Vfuncout;
    std::string __Vtask_broken_str_arg__1__Vfuncout;
    // Body
    VL_NULL_CHECK(vlSelf->tjo__DOT__unnamedblk1__DOT__tc, "tjo.sv", 6)->__VnoInFunc_broken_str_arg(vlSymsp, 0x666f6fU, __Vtask_broken_str_arg__0__Vfuncout);
    VL_WRITEF("%@\n",-1,&(__Vtask_broken_str_arg__0__Vfuncout));
    VL_NULL_CHECK(vlSelf->tjo__DOT__unnamedblk1__DOT__tc, "tjo.sv", 7)->__VnoInFunc_broken_str_arg(vlSymsp, 
                                                                                std::string("baar"), __Vtask_broken_str_arg__1__Vfuncout);
    VL_WRITEF("%@\n",-1,&(__Vtask_broken_str_arg__1__Vfuncout));
}
```

The interesting parts:
```cpp
    VL_NULL_CHECK(vlSelf->tjo__DOT__unnamedblk1__DOT__tc, "tjo.sv", 6)->__VnoInFunc_broken_str_arg(vlSymsp, 0x666f6fU, __Vtask_broken_str_arg__0__Vfuncout);
    ...
    VL_NULL_CHECK(vlSelf->tjo__DOT__unnamedblk1__DOT__tc, "tjo.sv", 7)->__VnoInFunc_broken_str_arg(vlSymsp, 
                                                                                std::string("baar"), __Vtask_broken_str_arg__1__Vfuncout);
```

wsnyder, 2022-08-16:
Did you try with today's master branch code?  I'd think yesterday's fix for #3544 would likely fix this.

Grulfen, 2022-08-16:
Yes. I tried with todays master. If i understand the fix for #3544 (f435d9624180c6b0d511f427ff41a876e3c7aaea) correctly it only handles string literals inside case statements

wsnyder, 2022-08-18:
Ah, right, something similar then.  If you'd like to take a look at V3Width to find the code that would be appreciated.

Grulfen, 2022-08-18:
I'm comparing the generated code from a TaskRef to a MethodCall, is it a sane assumption that they should be similar in this case? The test_regress example looks like this:
```systemverilog
package t_p;
    class T;
        function automatic void print_str(input string a_string);
            $display(a_string);
        endfunction
    endclass
endpackage

module t;

function automatic void display_str(input string b_string);
    $display(b_string);
endfunction

initial begin
    t_p::T t_c;
    t_c.print_str("print me");
    display_str("also me");
end
endmodule
```

The ...width.tree output contains this:
```
    1:2:2:1: METHODCALL 0x555556c08780 <e178#> {d21aj} @dt=0x555556cb0000@(w0)void  print_str pkg=0x555556be8180 -> TASK 0x555556c14000 <e9> {d3bh}  print_str [METHOD] [STMT] -> TASK 0x555556c14000 <e9> {d3bh}  print_str [METHOD]
    1:2:2:1:2: VARREF 0x555556bea6c0 <e177#> {d21af} @dt=0x555556bf0ea0@(w0)class:T  t_c [RV] <- VAR 0x555556be8780 <e176#> {d20am} @dt=0x555556bf0ea0@(w0)class:T  t_c [VSTATIC]  VAR
    1:2:2:1:3: ARG 0x555556bf0a90 <e127> {d21at}
    1:2:2:1:3:1: CONST 0x555556c08000 <e66> {d21at} @dt=0x555556bf09c0@(G/w64)  64'h7072696e74206d65
    1:2:2:1: TASKREF 0x555556c083c0 <e79> {d22af}  display_str -> TASK 0x555556c14200 <e40> {d15az}  display_str
    1:2:2:1:3: ARG 0x555556bf0c30 <e76> {d22ar}
    1:2:2:1:3:1: CVTPACKSTRING 0x555556c94420 <e188#> {d22ar} @dt=0x555556bf0340@(G/str)
    1:2:2:1:3:1:1: CONST 0x555556c08280 <e182#> {d22ar} @dt=0x555556bf0b60@(G/w56)  56'h616c736f206d65
```

So the TaskRef has converted the string literal to a CVTPACKSTRING but the MethodCall has not.

I think that the `visit(AstMethodCall nodep*)` function should be updated, but I'm having a hard time figuring out what needs to change. I'm guessing it is something on these lines: https://github.com/verilator/verilator/blob/db5fdfb0ee871e4cf8a25b8890207b507d5fa7c9/src/V3Width.cpp#L2569-L2571

For TaskRef the `processFTaskRefArgs` function is called, should something similar be done for MethodCall arguments as well?

wsnyder, 2022-09-07:
>I'm comparing the generated code from a TaskRef to a MethodCall, is it a sane assumption that they should be similar in this case? 

Correct.

>For TaskRef the processFTaskRefArgs function is called, should something similar be done for MethodCall arguments as well?

I think that's a right diagnosis of the fix.

wsnyder, 2022-09-11:
Thanks for your pull!

closed 2022-09-11 by wsnyder