verilator/verilator #3548: Dynamic array initialized with one element get converted to unsigned

Grulfen, 2022-08-16:
On verilator master branch

The following code:
```systemverilog
module tjo();

initial begin
    int some_ints[] = {12};
    $display("%d", some_ints[0]);
end

endmodule
```
Get converted to: 
```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Body
    VL_WRITEF("%11d\n",32,0xcU.at(0U));
}
```
Note the `0xcU.at(0U)` part.

This error does not happen for this code:
```systemverilog
module tjo();

initial begin
    int some_ints[] = '{12};
    $display("%d", some_ints[0]);
end

endmodule
```

Or this code:
```systemverilog
module tjo();

initial begin
    int some_ints[] = {12, 13};
    $display("%d", some_ints[0]);
end

endmodule
```

The working cases are converted to this or similiar:
```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Init
    VlQueue<IData/*31:0*/> tjo__DOT__unnamedblk1__DOT__some_ints;
    // Body
    tjo__DOT__unnamedblk1__DOT__some_ints = VlQueue<IData/*31:0*/>::cons(0xdU, 
                                                                         0xcU);
    VL_WRITEF("%11d\n",32,tjo__DOT__unnamedblk1__DOT__some_ints.at(0U));
}
```

I have tried with arrays of `int` and `string` and they both have this problem.

wsnyder, 2022-08-16:
Might you be willing to look into adding the test and making a pull request to fix this?

V3Width converts the unfortunate overloaded IEEE replicate/concat {} syntax to match the '{} syntax internally, so it is probably what botches this.

Grulfen, 2022-08-16:
Sure I can try to make a fix for this

closed 2023-02-01 by wsnyder