verilator/verilator #3549: Faster Verilog builds when Verilating many small modules sequentially?

gussmith23, 2022-08-16:
Hi all!

I have many small Verilog modules that I am simulating with Verilator. I've found that the process of verilating the files is taking much longer than any other part of the process (e.g. generating the Verilog modules themselves, or running the executable generated by Verilator). I'm building my code with a call to Verilator like:

```
verilator --cc --exe --build my_module.v my_testbench.cc
```

I haven't gathered hard data, but based on what's being printed to console when, it seems like most of  the time is spent in running the `ar` command.

I really need to speed up this part of my code, as it's taking unnecessarily long. Here are some of the approaches I've tried so far. I'd love help on them, because currently neither of them are working for me.

1. Calling Verilator just once. I thought I could cut out the overhead of running Verilator by calling it just once. I generated all of my small Verilog modules, and then generated a single .cc testbench to test all of the modules at once. However, I didn't realize that Verilator won't generate multiple top-level modules during a single invocation. I could try to group all of my modules into a larger module, but that instantly gets a lot more complicated!
2. Attempting to skip the `ar` step. It seemed like it was the `ar` step that took the longest, so I attempted to skip it. To do so, I tried essentially building the Verilated code myself. I would run Verilator with `verilator --cc my_module.v` to generate the .cpp and .h files, and then I would attempt to compile them together with my testbench with `clang++ my_module_0.cpp my_module_1.cpp ... my_testbench.cc`. However, I pretty quickly ran into issues, and realized that Verilator probably doesn't intend for its generated code to be very usable outside of its generated Makefiles. I had to make sure `verilated.h` was on the include path; I had to build `verilated.cpp`; after I did those things, I started to get even more hard-to-understand errors like 
   ```Undefined symbols for architecture x86_64:
  "std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::compare(unsigned long, unsigned long, char const*, unsigned long) const", referenced from:
      bool std::__1::operator==<char, std::__1::char_traits<char>, std::__1::allocator<char> >(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, char const*) in verilated-575a8b.o```

In short, I'm trying to Verilate many small Verilog modules quickly, but the standard Verilator flow is too slow, and I can't get any alternative Verilator flow to work. Any suggestions or help?

gussmith23, 2022-08-16:
Ah, I spoke too soon -- the errors using the second method seem to be MacOS build related. I fixed them by building with `cc -lstdc++ -stdlib=libc++ ...`

Still, I'm wondering if this second route is advisable, or if it's considered a misuse of Verilator? I'm also open to any better suggestions!

wsnyder, 2022-08-18:
The Apple "ar" command has been a source of much pain and is not very good, if you can avoid Mac you'll have a better experience, otherwise it should be reasonable to build Verilator outside the standard Makefile.  If you get any undefined symbols you are likely missing some library or .o file - from your message maybe libc++.  If you look at the ar and g++ commands that Verilator's defaults create that should give you an idea of what is needed.

Closing as answered but feel free to follow up.

closed 2022-08-18 by wsnyder