verilator/verilator #3554: Getting unexpectedly "Warning-REALCVT: Implicit conversion of real to integer", but only for certain values

NilsGraf, 2022-08-17:
Only for certain values, I'm getting the following warning unexpectedly:
``Warning-REALCVT: Implicit conversion of real to integer``

I'm using the following version (on MacBook Air, which I installed via brew):
``Verilator 4.222 2022-05-02 rev UNKNOWN.REV``

Example code (this code is adopted from [here](https://www.chipverify.com/systemverilog/systemverilog-datatypes)):
```verilog
struct packed {
   int	coins;
   real	dollars;
} wallet;

initial begin
  wallet = '{5, 0.0};
  wallet = '{5, 2.0};
  wallet = '{5, -2.0};
  wallet = '{5, 2.5};
end
```

Verilator is ok with ``0.0`` and ``2.0``, but not with ``-2.0`` and ``2.5``, see below two warnings:
```
%Warning-REALCVT: bla.sv:22:17: Implicit conversion of real to integer
   22 |   wallet = '{5, -2.0};
      |                 ^
%Warning-REALCVT: bla.sv:23:17: Implicit conversion of real to integer
   23 |   wallet = '{5, 2.5};
      |                 ^~~
```

Any ideas?  Thank you!

wsnyder, 2022-08-18:
Thanks for the good issue report.

As with #3510 these are side effects of #1541, and likely can't be fixed until that one is.

closed 2022-08-18 by wsnyder

NilsGraf, 2022-08-18:
Hi Wilson, thank you for the prompt response. What's the approx ETA for fixing #1541 ?

wsnyder, 2022-08-18:
Probably a year as no one is working on it and it is significant effort. Perhaps you would?

NilsGraf, 2022-08-18:
No worries, I found a workaround using ``$realtobits()`` and ``$bitstoreal()``, see the modified code below. I wonder if Verilator could implement this same workaround under the hood (i.e. convert ``real`` in structs to bits using ``$realtobits()`` and then ``$bitstoreal()`` when the struct is read)? 

```verilog
struct packed {
   int		coins;
   logic [63:0]	dollars;
} wallet;

initial begin
  wallet = '{5, $realtobits(0.0)};
  wallet = '{5, $realtobits(2.0)};
  wallet = '{5, $realtobits(-2.0)};
  wallet = '{5, $realtobits(2.5)};

  $display("%.2f", $bitstoreal(wallet.dollars));
end
```