verilator/verilator #3558: Support shortreal

NilsGraf, 2022-08-18:
Any plans to support ``shortreal``?

Here is the warning you get when using ``shortreal``
```
%Warning-SHORTREAL: test.sv:60:1: Unsupported: shortreal being promoted to real (suggest use real instead)
```

wsnyder, 2022-08-20:
Note you can just turn off that warning.

Many other simulators likewise convert shortreal to real, so what does shortreal do for you that real does not?

NilsGraf, 2022-08-20:
Wouldn't shortreal only occupy 32-bit instead of 64-bit? So perhaps this would reduce the memory footprint for large arrays of shortreals?

wsnyder, 2022-08-20:
Potentially, but that's something worth considering if we have data showing that is causing a performance impact on a design that uses shortreal a lot.

wsnyder, 2022-08-27:
Unlikely shortreal is worth the initial and ongoing support cost m, especially as several closed source simulators likewise alias to real.

closed 2022-08-27 by wsnyder