verilator/verilator #3567: Comparison with negation doesn't work

RRozak, 2022-08-22:
The following test fails on current master:
```verilog
module t (/*AUTOARG*/
   );
   wire a = 1;

   wire b = 0 == ~a;

   always begin
      if (b) begin
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end
endmodule
```
Here is AST of assignments (from *_012_width.tree):
```
    1:2: ASSIGNW 0x555556bc0000 <e91#> {c9al} @dt=0x555556b2a270@(w1)
    1:2:1: CONST 0x555556bbe3c0 <e104#> {c9an} @dt=0x555556b2a270@(w1)  1'h1
    1:2:2: VARREF 0x555556b24240 <e90#> {c9aj} @dt=0x555556b2a270@(w1)  a [LV] => VAR 0x555556b22180 <e89#> {c9aj} @dt=0x555556b2a270@(w1)  a [VSTATIC]  WIRE
    1:2: VAR 0x555556b22300 <e107#> {c11aj} @dt=0x555556b2a680@(w1)  b [VSTATIC]  WIRE
    1:2: ASSIGNW 0x555556bc0210 <e109#> {c11al} @dt=0x555556b2a680@(w1)
    1:2:1: EQ 0x555556bc0160 <e45> {c11ap} @dt=0x555556b2a5b0@(G/w1)
    1:2:1:1: CONST 0x555556bbe140 <e35> {c11an} @dt=0x555556b2a1a0@(G/swu32/1)  ?32?sh0
    1:2:1:2: NOT 0x555556bc00b0 <e115#> {c11as} @dt=0x555556b2ab60@(G/wu32/1)
    1:2:1:2:1: EXTEND 0x555556bc04d0 <e121#> {c11at} @dt=0x555556b2ab60@(G/wu32/1)
    1:2:1:2:1:1: VARREF 0x555556b24480 <e119#> {c11at} @dt=0x555556b2a270@(w1)  a [RV] <- VAR 0x555556b22180 <e89#> {c9aj} @dt=0x555556b2a270@(w1)  a [VSTATIC]  WIRE
```
It looks like the problem is that `a` (which equals `1'h1`) is first extended to 32 bits, so it equals `32'b0...001` then. Negation flips these bits and then it equals `32'b1...110`, so comparison with `0` returns false.

wsnyder, 2022-08-22:
I believe thebehavior is correct as 0 is 32bits (use 1'b0 otherwise). Does it work on other aimulators?

RRozak, 2022-08-22:
I've checked on Aldec Riviera on https://edaplayground.com/ and behavior is the same as on Verilator.
I haven't found in specs how such comparison should be handled. I thought that we should extend the result of negation, not the constant under negation.

wsnyder, 2022-08-28:
The == operator is first, and it takes the wider of its two operands.

closed 2022-08-28 by wsnyder