verilator/verilator #3571: Separate Unused parameters from Unused signals

topatopa, 2022-08-23:
I recently updated my verilator and noticed this change (https://github.com/verilator/verilator/issues/2627). I know it has been discussed in here (https://github.com/verilator/verilator/issues/2798), but I want to give a second opinion.

I like that in verilator, you can quickly check for errors in a file by running `verilator --lint-only -Wall "filename1.sv"`. I find it the fastest way to fix mistakes in code without having to lint the code. Unused signals are usually stuff not properly wired, while unused parameters is rarely a mistake. For example, you set up the parameters for a standardized interface, but you are only using a few parameters. Now you will get a warning for each unused parameter. 

I think it would be great to be able to do something like this `verilator --lint-only -Wall -Wno-unused-parameter "filename1.sv"` to fix the signals without having to look at the parameters.

wsnyder, 2022-08-25:
Makes sense, perhaps make UNUSEDVAR, UNUSEDPARAM? Should it be even finer split? For backwards compatibility we should suppress both warnings on UNUSED.

Would you be willing to make a pull request for this?

topatopa, 2022-08-27:
I think that would be great. I don't have much experience with verilator development, but I can probably do it from the commit that added unused parameters (https://github.com/verilator/verilator/commit/47eeef485d1781f104211d96ffb7498f55ae1b2b). I have some time later next week.

closed 2022-10-18 by topatopa