verilator/verilator #3573: Q: Issue with compiling and running Verilator SystemC tests

mslijepc, 2022-08-24:
Hi, how should we `configure`/`install` Verilator to be able to compile and run SystemC tests?

I used the following command to configure Verilator
```
sudo ./configure --prefix=/work/verilator/`git describe | sed "s/verilator_//"` CXX="clang++ -std=c++14" LINK=clang++
```
During the config, I get the following output
```
checking whether SystemC is found (in system path)... no
```
But, SystemC variables are set
```
env | grep SYSTEMC

SYSTEMC_HOME=/work/systemc-2.3.3
SYSTEMC_LIBDIR=/work/systemc-2.3.3/lib-linux64
SYSTEMC_INCLUDE=/work/systemc-2.3.3/include
```
`make` and `make test` run without any problem.

If I try to execute some SystemC test I get the following warning:
```
-Skip: dist/t_var_sc_bv: scenario 'dist' not enabled for test
dist/t_var_sc_bv: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
vlt/t_var_sc_bv: ==================================================
-Skip: vlt/t_var_sc_bv: Test requires SystemC; ignore error since not installed
```

Should I add some flag during the `configure` (i.e. SystemC path as `CXX` or something like that) or I did some other step wrong?

Thank you

wsnyder, 2022-08-28:
configure checks if systemc is installed as a system library (e.g. "apt-get install libsystemc"), if you look at configure.ac you'll see it tries to compile a program using sc_main.
If you don't have it installed, then configure won't find it, and you will need to set SYSTEMC_INCLUDE (at least) before you run "make test" or any use of Verilator using SystemC.

closed 2022-08-28 by wsnyder

drom, 2022-08-29:
I have very similar problem. I have local installation of SystemC and setup `SYSTEMC_INCLUDE` before `make test` and getting the same error:

```
clang++ -std=c++14     -L/sifive/systemc-2.3.3/lib64-linux64 Vt_a2_first_sc__main.o verilated.o verilated_vcd_c.o verilated_vcd_sc.o Vt_a2_first_sc__ALL.a     -lsystemc -o Vt_a2_first_sc
driver: Leaving directory '/home/drom/work/github/verilator/verilator/test_regress/obj_vlt/t_a2_first_sc'
        obj_vlt/t_a2_first_sc/Vt_a2_first_sc    > obj_vlt/t_a2_first_sc/vlt_sim.log
obj_vlt/t_a2_first_sc/Vt_a2_first_sc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory
%Warning: vlt/t_a2_first_sc: Exec of obj_vlt/t_a2_first_sc/Vt_a2_first_sc failed: obj_vlt/t_a2_first_sc/Vt_a2_first_sc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory

vlt/t_a2_first_sc: %Error: Exec of obj_vlt/t_a2_first_sc/Vt_a2_first_sc failed: obj_vlt/t_a2_first_sc/Vt_a2_first_sc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory
vlt/t_a2_first_sc: FAILED: Exec of obj_vlt/t_a2_first_sc/Vt_a2_first_sc failed: obj_vlt/t_a2_first_sc/Vt_a2_first_sc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:03
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:03

======================================================================
        #vlt/t_a2_first_sc: %Error: Exec of obj_vlt/t_a2_first_sc/Vt_a2_first_sc failed: obj_vlt/t_a2_first_sc/Vt_a2_first_sc: error while loading shared libraries: libsystemc-2.3.3.so: cannot open shared object file: No such file or directory
                make && test_regress/t/t_a2_first_sc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:03
```

mslijepc, 2022-08-30:
I am using _openSUSE Tumbleweed_, so `apt-get install libsystemc` will not work. SystemC is installed on my computer, and `SYSTEMC_INCLUDE` is set. I even tried to install it in `/usr/local`, but no success.

wsnyder, 2022-08-30:
If compiling this doesn't work, it isn't installed in the system-standard location, and you will need SYSTEMC_INCLUDE.  Note setting SYSTEMC_INCLUDE configure will still say systemc is not present, that is ok, you must instead manually set the SYSTEMC_INCLUDE and SYSTEMC_LIB each compile.

      #include <systemc.h>
      extern "C" int sc_main(int argc, char* argv[]) { return 0; }