verilator/verilator #3576: v5_develop / Timing support / wrong VCD/FST traces

mathieu-tp, 2022-08-25:
This version of dynamic scheduler as still a few problems with the VCD or FST dump for waveform display.

With the attached verilog and main CC files
[test_dynamic.zip](https://github.com/verilator/verilator/files/9424106/test_dynamic.zip)

The generated vcd file seems good, timings have exactly the expected values but:

- the main "clk" signal is stuck at 0, while the several copies (either by an explicit delay or an event) have the good shape.

- the VCD file contains a lot of "empty" time stamps linked (i presume) to the way the scheduling if done. It should be preferable to generate time stamps only when there is real activity on signals.

kbieganski, 2022-08-25:
https://github.com/verilator/verilator/pull/3579 should fix the first problem.

The second one could be fixed by lazy-printing the timestamp, but at its core it's really a performance issue that needs requires some more thought. I will open a draft PR that fixes it (in some cases) soon.

wsnyder, 2022-10-16:
#3678 seems similar so should be considered as part of this issue, that's problem is __Vm_traceActivity which I suspect is also part of this issue's problem.

closed 2022-10-20 by wsnyder