verilator/verilator #3583: hier_block mode module has wrong port connection order

Kelin9298, 2022-08-29:
When I compile the below case with `verilator -cc dut.v --hierarchical --debug` (verilator version: 4.225)
### dut.v
```
`timescale 1ps/1ps

module mod2 (
    output reg q,
    input wire d,
    input wire iclk
);
    
    /*verilator hier_block*/
    
    always@(posedge iclk)
        q = d;
    
endmodule

module dut (
    output wire q, 
    input wire d,
    input wire clk
);

    mod2 u_mod (q, d, clk);

endmodule

```
Will get error message:
```
%Error-ASSIGNIN: dut.v:21:20: Assigning to input/const variable: 'd'
-node: VARREF 0xf01420 <e519#> {e21au} @dt=0@  d [LV] => VAR 0xf17620 <e355> {e17aq} @dt=0@  d INPUT [VSTATIC]  WIRE
   21 |     mod2 u_mod (q, d, clk);
      |                    ^
                 ... For error description see https://verilator.org/warn/ASSIGNIN?v=4.225
```
Here's what I found in a debug message file:
### Vdut_007_linkjump.tree
```
(omit)
1:2: CELL 0x2a4a620 <e376> {e21ak}  u_mod -> MODULE 0x2a3a540 <e400#> {d6ai}  mod2  L3 [1ps]
    1:2:1: PIN 0x2a49ff0 <e365> {e21ar}  __pinNumber1 -> VAR 0x2a273e0 <e13> {d7av} @dt=0@  iclk INPUT [VSTATIC]  PORT
    1:2:1:1: VARREF 0x2a33310 <e516#> {e21ar} @dt=0@  q [RV] <- VAR 0x2a49100 <e349> {e16ar} @dt=0@  q OUTPUT [VSTATIC]  WIRE
    1:2:1: PIN 0x2a4a290 <e370> {e21au}  __pinNumber2 -> VAR 0x2a27950 <e21> {d8ay} @dt=0@  q OUTPUT [VSTATIC]  PORT
    1:2:1:1: VARREF 0x2a33420 <e519#> {e21au} @dt=0@  d [LV] => VAR 0x2a49620 <e355> {e17aq} @dt=0@  d INPUT [VSTATIC]  WIRE
    1:2:1: PIN 0x2a4a530 <e374> {e21ax}  __pinNumber3 -> VAR 0x2a27e70 <e30> {d9ax} @dt=0@  d INPUT [VSTATIC]  PORT
    1:2:1:1: VARREF 0x2a33530 <e522#> {e21ax} @dt=0@  clk [RV] <- VAR 0x2a49b40 <e362> {e18aq} @dt=0@  clk INPUT [VSTATIC]  WIRE
(omit)
```
You can see that the pin order is different from the order in the dut.v. In the Vdut_007_linkjump.tree, the mod2's pin order is `iclk -> q -> d`, but in the dut.v the mod2's pin order is `q -> d -> iclk`. I think it's the cause of the ASSIGNIN error.

I also found that if I change to put the `d` in the always block like `always@(posedge d)` instead of `always@(posedge iclk)`, the `__pinNumber1`  will be `d` instead of `iclk`.

Is this a bug or is that I have to add some options to prevent this error? Thanks

wsnyder, 2022-08-30:
Thanks for the good problem report, the order of variables was changing internally, fixed.

closed 2022-08-30 by wsnyder

Kelin9298, 2022-08-30:
Hi @wsnyder,

Thanks for the fix, I replaced my V3VariableOrder.cpp with the fixed one and rebuild my verilator, and the **dut.v** case seems correct now. But after I run the below **dut3.v** case with `verilator -cc dut3.v --hierarchical --debug`.
### dut3.v
```
`timescale 1ns/1ns
module mod3(out3, in3, clock3, reset3)
    output  out3;
    input   in3, reset3, clock3;
    reg     out3;

    /*verilator hier_block*/

    always@(posedge clock3)
        out3 <= in3
    
endmodule

module dut3(out, in, clock, reset)
    input   reset, clock, in;
    output  out;
    wire    reset, clock;
    wire    out;

    mod3 u_mod3(out, in, clock, reset)

endmodule
```
In the debug file:
### Vdut3_007_linkjump.tree
```
(omit)
1:2: CELL 0x2004440 <e446> {e20ak}  u_mod3 -> MODULE 0x1fe1050 <e476#> {d6ai}  mod3  L3 [1ns]
1:2:1: PIN 0x1feb890 <e431> {e20ar}  __pinNumber1 -> VAR 0x1fe1680 <e13> {d7aw} @dt=0@  out3 OUTPUT [VSTATIC]  PORT
1:2:1:1: VARREF 0x1fded40 <e628#> {e20ar} @dt=0@  out [LV] => VAR 0x2002d10 <e412> {e16an} @dt=0@  out OUTPUT [VSTATIC]  WIRE
1:2:1: PIN 0x2003de0 <e436> {e20aw}  __pinNumber2 -> VAR 0x1fe1c00 <e21> {d8ax} @dt=0@  in3 INPUT [VSTATIC]  PORT
1:2:1:1: VARREF 0x1fdee50 <e631#> {e20aw} @dt=0@  in [RV] <- VAR 0x20029b0 <e407> {e15bb} @dt=0@  in INPUT [VSTATIC]  PORT
1:2:1: PIN 0x2004090 <e440> {e20ba}  __pinNumber3 -> VAR 0x1fe2130 <e30> {d9ax} @dt=0@  reset3 INPUT [VSTATIC]  PORT
1:2:1:1: VARREF 0x1fdef60 <e634#> {e20ba} @dt=0@  clock [RV] <- VAR 0x20026a0 <e402> {e15au} @dt=0@  clock INPUT [VSTATIC]  WIRE
1:2:1: PIN 0x2004340 <e444> {e20bh}  __pinNumber4 -> VAR 0x1fe2660 <e39> {d10ax} @dt=0@  clock3 INPUT [VSTATIC]  PORT
1:2:1:1: VARREF 0x1fdf070 <e637#> {e20bh} @dt=0@  reset [RV] <- VAR 0x2002390 <e447> {e15an} @dt=0@  reset INPUT [VSTATIC]  WIRE
(omit)
```
You can see that the order of **PIN** is `reset3 -> clock3` which is not the same as the order of the `mod3` port declaration, which is `clock3 -> reset3`. I guess this is considered as not correct? Thanks

wsnyder, 2022-08-30:
Can you file a separate bug on that one please?

Kelin9298, 2022-08-30:
Hi @wsnyder , 
I've filed a new bug at #3585 . Thanks!