verilator/verilator #3584: public flat comment causing model constructor segfault

lwshowl, 2022-08-30:
im trying to co-sim with a c++ axi slave device , the c++ slave model requires to read verilated axi interfaces in c++ 
i have written an axi master , but since the axi master is not module , i cant access the axi intefaces after verilating it
i read the document and append /*verilator public flat*/ after the interface wires , then i can access the interface signals. but after doing that , when initilize the verilated model segmentfault is immediately triggered , i backtraced the error and here is what i found (pasted following), it seems that the constructor of the verilated model has caused the trouble.

if there is no public flat comment , the code works just fine.

#0  0x00007ffff7e7225e in std::_Rb_tree_decrement(std::_Rb_tree_node_base*) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x000055555556d4e7 in std::_Rb_tree_iterator<std::pair<char const* const, VerilatedScope const*> >::operator-- (
    this=0x7fffffffda30)
    at /usr/include/c++/9/bits/stl_tree.h:302
#2  0x000055555556d429 in std::_Rb_tree<char const*, std::pair<char const* const, VerilatedScope const*>, std::_Select1st<std::pair<char const* const, VerilatedScope const*> >, VerilatedCStrCmp, std::allocator<std::pair<char const* const, VerilatedScope const*> > >::_M_get_insert_unique_pos (
    this=0x5555555bc3a0 <VerilatedImp::s_s+96>,
    __k=@0x5555555e36f0: 0x5555555e3710 "TOP.core")
    at /usr/include/c++/9/bits/stl_tree.h:2104
#3  0x000055555556a9b4 in std::_Rb_tree<char const*, std::pair<char const* const, VerilatedScope const*>, std::_Select1st<std::pair<char const* const, VerilatedScope const*> >, VerilatedCStrCmp, std::allocator<std::pair<char const* const, VerilatedScope const*> > >::_M_get_insert_hint_unique_pos (
    this=0x5555555bc3a0 <VerilatedImp::s_s+96>, __position=
  {first = 0x0, second = 0x0},
    __k=@0x5555555e36f0: 0x5555555e3710 "TOP.core")
    at /usr/include/c++/9/bits/stl_tree.h:2201
#4  0x0000555555568b8a in std::_Rb_tree<char const*, std::pair<char const* const, VerilatedScope const*>, std::_Select1st<std::pair<char const* const, VerilatedScope const*> >, VerilatedCStrCmp, std::allocator<std::pair<char const* const, VerilatedScope const*> > >::_M_emplace_hint_unique<std::pair<char const*, VerilatedScope const*> > (this=0x5555555bc3a0 <VerilatedImp::s_s+96>, __pos=
  {first = 0x0, second = 0x0})
    at /usr/include/c++/9/bits/stl_tree.h:2459
#5  0x00005555555672c1 in std::map<char const*, VerilatedScope const*, VerilatedCStrCmp, std::allocator<std::pair<char const* const, VerilatedScope const*> > >::insert<std::pair<char const*, VerilatedScope const*> > (this=0x5555555bc3a0 <VerilatedImp::s_s+96>,
    __position={first = 0x0, second = 0x0}, __x=...)
    at /usr/include/c++/9/bits/stl_map.h:876
#6  0x0000555555565ac5 in VerilatedImp::scopeInsert (
    scopep=0x5555555e3618)
    at /usr/share/verilator/include/verilated_imp.h:299
--Type <RET> for more, q to quit, c to continue without paging--c
#7  0x0000555555563e06 in VerilatedScope::configure (this=0x5555555e3618, symsp=0x5555555e3600, prefixp=0x5555555e35b0 "TOP", suffixp=0x5555555a57d3 "core", identifier=0x5555555a57d3 "core", type=VerilatedScope::SCOPE_OTHER) at /usr/share/verilator/include/verilated.cpp:2328
#8  0x0000555555592afc in Vcore__Syms::Vcore__Syms (this=0x5555555e3600, topp=0x5555555d9f00, namep=0x5555555e35b0 "TOP") at Vcore__Syms.cpp:23
#9  0x000055555557e9a9 in Vcore::Vcore (this=0x5555555d9f00, __VCname=0x5555555a4010 "TOP") at Vcore.cpp:13

wsnyder, 2022-08-30:
Can you please attach a self contained test case showing this, ideally in test_regress format (see https://github.com/verilator/verilator/blob/master/docs/internals.rst)?

I'm not sure if public inside interfaces has been tried before, so this might be complicated to fix.

lwshowl, 2022-08-31:
> Can you please attach a self contained test case showing this, ideally in test_regress format (see https://github.com/verilator/verilator/blob/master/docs/internals.rst)?
> 
> I'm not sure if public inside interfaces has been tried before, so this might be complicated to fix.
thanks!

for my case ,maybe its more reasonable to put axi signals at top level , maybe i should not hide axi wires inside submodules ?
i think the only thing special is the public flat comment itslef , i never ran into such problems without the public flat comment.

i also found out that , if the public comment is added inside a submodule , the constructor in the syms file generates one more  “configure" function(in total of 2) of the submodule class which seemingly sets the names and scopes for the submodule class. if there is no public flat comment, there is only one configure function of top module class.

wsnyder, 2022-08-31:
It is much better to not use public, and instead either route all signals directly, or make a wrapper that has the additional outputs you want and looks at what you need using dotted notation (e.g. "output view; assign view = look.into.sub.module.view;").

closed 2022-09-26 by wsnyder