verilator/verilator #3588: Optional override C++ std version in verilator.mk

davidbiancolin, 2022-09-01:
I'd like to verilate designs with a bunch of C++ sources monolithically using >= C++17, but autoconf on my platform produces a `verilated.mk` that appends to gnu++14 (since i'm not on arch linux or a new enough version of ubuntu).  

I've read the discussion here: https://github.com/verilator/verilator/issues/3081. In my case i have built my own compiler and glibc from source (well, pulled them in using conda), and i've tried manually changing `verilated.mk` to use the standard library version i want successfully. 

So, would a PR changing `verilated.mk` to accept a variable that'd take precedence over `CFG_CXXFLAGS_STD_NEWEST` be welcome, or considered too dangerous? Alternatively, if there was another way of changing the default stdlib version either at installation time or at make invocation time that would be better received i'd be happy to implement it.  

Thanks for making + supporting verilator. :)

wsnyder, 2022-09-01:
I would have thought that "./configure CXX=my_gcc_version" by running the new version would see it is compatible and not add the flags.  Does that work?

wsnyder, 2022-09-26:
Can you comment if "./configure CXX=my_gcc_version" worked?

davidbiancolin, 2022-09-27:
I'm using gcc 10.3.0, and it did not work, but i was mistaken when i said "i had built my own glibc". That is not being provided in my conda environment as i had suggested, and instead is using the ubuntu 18.04 default installation. 

I need to see what's happening in the conda-forge build container for verilator and get back to you.

timsnyder, 2022-09-28:
Hey @wsnyder.  We `./configure` with `CXX` set in the environment and I see configure successfully running the version of g++ that I would expect in the build logs at https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=561395&view=logs&j=656edd35-690f-5c53-9ba3-09c10d0bea97&t=e5c8ab1d-8ff9-5cae-b332-e15ae582ed2d&l=221

>  checking whether $BUILD_PREFIX/bin/x86_64-conda-linux-gnu-c++ accepts -std=gnu++14... yes

If I am understanding the logic in https://github.com/verilator/verilator/blob/a999c73ce080c08461d994897c06bd1d7a4bd890/configure.ac#L349-L367 correctly:
* configure will always set `CFG_CXXFLAGS_STD_NEWEST` to the first `-std` option that is accepted.  I don't see it determining that the new version is compatible and not adding a `-std` option.
* `c++17` (or `gnu++17`) will only be attempted if `lsb_release` says `*Arch*Linux* | *Ubuntu*22.04*`.  I don't understand why you wouldn't just try the c++17 standard regardless of the platform reported by `lsb_release`.

@wsnyder You guys have the `CFG_CXXFLAGS_STD_NEWEST` in verilated.mk to ensure that the build-time (for both verilator and SystemC) and verilation-time std are the same?  Is this to avoid ABI differences that cause linking errors? 

@davidbiancolin I'm fine forcing the conda-forge build of Verilator to use `-std=c++17` to work around this for now.  I've opened a PR to do so in conda-forge/verilator-feedstock#25.  The conda-forge compilers (recent versions at least) all support c++17 with no problem.

wsnyder, 2022-09-28:
Ideally configure would just pick the "newest" --std that works (hence the name of the variable).  We used to do that and it caused breakage; the problem is that glibc does not support arbitrary choices of --std, neither newer or older and generally just one or two choices will work at all.  It's not an easy thing to determine in configure either (AFAIK).

Looking at the distro is certainly a hack.

I suspect we will need to require that when the user sets CC to something nonstandard they also override this logic by also explicitly passing CFG_CXXFLAGS_STD_NEWEST or perhaps some better named new variable.

gezalore, 2022-09-29:
The reason we look for specific C++ versions on specific distros is to match the C++ version the distro SystemC library was built with, as we need to match if we want to use them (which we do by default, at least for testing). Improvements are welcome.

timsnyder, 2022-09-29:
Thanks @wsnyder and @gezalore.  Currently for the conda-forge verilator package, I build without SystemC support (because no one has asked me to add it and I don't need it).  If/when it is added, I'll make sure that the Verilator package matches.  This might be a good time for me to add the SystemC package to conda-forge if only so that I'm in the loop and can define the testing relationship with Verilator...

I think the conda-forge recipe should probably just pass `CFG_CXXFLAGS_STD_NEWEST` to `configure` with a comment documenting why it is set.  The logic you have in the configure script makes complete sense for using the distro packages and I don't expect you guys to have packager-specific logic correct in all use cases.

Just FYI (and for my remembering) the conda-forge package recipes have explicitly added `-std=c++XX` to `CXXFLAGS` for many years and are moving away from that to using the default std of the compiler.  Actually, looking back through [some things](https://github.com/conda-forge/ctng-compiler-activation-feedstock/pull/50/files#diff-d7075654874cb08007a21aaab3ecd4b3453a9087e7505d034d548b8938b599bcR45-R52) I see that once we're using gcc 11 to build packages, `-std` will no longer be included in `CXXFLAGS`.  In my latest builds, I see that gcc 10 is still being used.

@davidbiancolin do you still see a need for being able to override `CFG_CXXFLAGS_STD_NEWEST` at verilator runtime?  I think we may just be able to add `CFG_CXXFLAGS_STD_NEWEST=` to the make cmdline at https://github.com/firesim/firesim/blob/230314a4d7a4bf1c215389a9836a8cc8767a0ccd/sim/midas/src/main/cc/rtlsim/Makefrag-verilator#L42 to override it now.

davidbiancolin, 2022-09-29:
I think i'd still prefer to be able to override the C++ version at design-compile/verilation time, and let the user fall on their own sword if they get it wrong. However, if providing a value at configure time is desired, i can PR that.

When we can't use conda, i don't think it'll be too onerous to have products of verilator version X C++ library versions installed.

mithro, 2022-09-30:
FYI - @proppy

mithro, 2022-09-30:
FYI - @PiotrZierhoffer

wsnyder, 2022-10-15:
Is there some change to Verilator needed for this, if so what?

davidbiancolin, 2022-10-17:
Essentially change the assignment here https://github.com/verilator/verilator/blob/master/include/verilated.mk.in#L25 to `?=`.  (or something equivalent, like define `CFG_CXXFLAGS_STD_OVERRIDE` that takes precedence when set by the user. 

If that's sensible i can open a PR, i'll go ahead an close this issue.

timsnyder, 2022-12-01:
@wsnyder , David and I talked more and agree that we can override the behavior with `-MAKEFLAGS CFG_CXXFLAGS_STD_NEWEST=` (i.e. setting it empty) so that the `-std` we request in `-CFLAGS` is used.

We can close this issue.

closed 2022-12-01 by wsnyder