verilator/verilator #3589: Towards constrained randomization

kbieganski, 2022-09-02:
We are looking into adding constrained randomization support to Verilator. Obviously that is a pretty big undertaking, so we would like to reuse some existing work for this. It appears there are only two solutions available for C/C++:
* [SCV from Accellera](https://www.accellera.org/activities/working-groups/systemc-verification), seems to require SystemC which is pretty much a deal-breaker,
* [CRAVE from the University of Bremen](https://github.com/agra-uni-bremen/crave), MIT license, looks to be well thought-out. It can use many solvers as backends, which is really cool. Unfortunately it doesn't work on a newer compiler without some tinkering, and isn't really active. So we would have to resurrect it and make it more compatible. It also has a bunch of dependencies, mostly Boost libs. However, looking at it closer, a lot of this Boost stuff could be replaced by C++11/14/17/20 equivalents, or reimplemented pretty quickly by hand. Their constraint DSL uses Boost.Proto, but we don't need that for Verilator, so we could remove it or put it behind some `#ifdef`s.

Of course, such a library would only be a dependency for those that actually use constrained random in their tests.

Would be great to hear your thoughts on this @wsnyder. Are you open to using something like this? Perhaps there's a different, better option?

wsnyder, 2022-09-02:
scv seems fairly small and uses CUDD under the covers.  I'd suggest look more into that as a starting point for our own C++ library, versus a C++ library we can directly reuse without changes.  The big benefit of scv is we know it has succeeded in solving the "same" problem, and is C++ based.

wsnyder, 2022-09-02:
Note scv is using cudd 2.3.0 which is old, if you looking into it use the latest (3.0.0?).

kbieganski, 2022-09-05:
So CRAVE is out of the question?

wsnyder, 2022-09-05:
I'm not ruling any options out at this point, it just seemed you had ruled out SCV and my point was the main library isn't SCV but cudd which is portable.

Though, FWIW I would consider requiring boost a major negative in the scorecarding of options.

kbieganski, 2022-09-05:
Oh I agree. That's why we're considering ripping it out. What I like about CRAVE is that it has multiple backends, including CUDD. If not CRAVE, perhaps we could use their solver wrapper library, metaSMT.

closed 2022-12-16 by wsnyder

InputBlackBoxOutput, 2024-01-08:
Hi @wsnyder & @kbieganski, What is the current status for constraint randomization. Is this begin worked on?

kbieganski, 2024-01-08:
I've been doing some research on it, but it's not a priority. I wouldn't expect a usable implementation anytime soon. Contributions are welcome :)