verilator/verilator #3590: Case eq statement doesn't work when net is compared with z

RRozak, 2022-09-02:
The following test fails currently on master:
```systemverilog
module t (input wire clk);
   wire a;
   assign a = clk;

   always begin
      if (a === 1'z)
         $stop;

      if (a == clk) begin
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule
```
The problem is that comparison `a === clk` returns true. If we replace reference to input port with constant, it works then properly.

gezalore, 2022-09-02:
There is no `a === clk` in the test case above, please edit for consistency.

RRozak, 2022-09-02:
Thanks, you're right. In the description I meant `a === 1'z`.

gezalore, 2022-09-02:
Also, `1'z` is "cast variable `z` (which does not exist) to 1-bit wide", does Verilator accept this test case as is?

RRozak, 2022-09-02:
Yes, it accepts.
Casting would be if I wrote `1'(z)`.

RRozak, 2022-09-02:
I changed it to `1'bz` to be sure that it is a proper constant according to standard. It didn't change the behavior.

gezalore, 2022-09-02:
> Casting would be if I wrote `1'(z)`.
Fair, still accepting `1'z` is a bug in itself.

wsnyder, 2022-09-05:
I suspect V3Tristate is botching this since the nets are non-tristate nets.  Perhaps you'd be willing to debug this with assistance?

RRozak, 2022-09-06:
Yes. It looks that you're right.
_023_begin.tree looks accurate. The problem is in _024_tristate.tree.
Let's use simplified code from the description to investigation:
```systemverilog
module t (input wire clk);
   wire a;
   assign a = clk;

   always begin
      if (a === 1'bz)
         $stop;
   end

endmodule
```
Here is _024_tristate tree:
```
     NETLIST 0x555556af8000 <e1> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x555556b025a0 <e95> {c1ai}  __024root  L1 [P] [1ps]
    1:2: CELL 0x555556b4c140 <e98> {c1ai}  t -> MODULE 0x555556b02120 <e97> {c1ai}  t  L0 [1ps]
    1:2:1: PIN 0x555556af81e0 <e102> {c1aw}  clk -> VAR 0x555556b00180 <e75> {c1aw} @dt=0x555556b08680@(G/w1)  clk INPUT [VSTATIC]  WIRE
    1:2:1:1: VARREF 0x555556b026c0 <e103> {c1aw} @dt=0x555556b08680@(G/w1)  clk [RV] <- VAR 0x555556b00480 <e99> {c1aw} @dt=0x555556b08680@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  WIRE
    1:2: VAR 0x555556b00480 <e99> {c1aw} @dt=0x555556b08680@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  WIRE
    1: MODULE 0x555556b02120 <e97> {c1ai}  t  L0 [1ps]
    1:2: VAR 0x555556b00180 <e75> {c1aw} @dt=0x555556b08680@(G/w1)  clk INPUT [VSTATIC]  WIRE
    1:2: VAR 0x555556b00300 <e91> {c2aj} @dt=0x555556b08680@(G/w1)  a [VSTATIC]  WIRE
    1:2: ASSIGNW 0x555556b4a000 <e93> {c3an} @dt=0x555556b08680@(G/w1)
    1:2:1: VARREF 0x555556b02240 <e81> {c3ap} @dt=0x555556b08680@(G/w1)  clk [RV] <- VAR 0x555556b00180 <e75> {c1aw} @dt=0x555556b08680@(G/w1)  clk INPUT [VSTATIC]  WIRE
    1:2:2: VARREF 0x555556b02360 <e92> {c3al} @dt=0x555556b08680@(G/w1)  a [LV] => VAR 0x555556b00300 <e91> {c2aj} @dt=0x555556b08680@(G/w1)  a [VSTATIC]  WIRE
    1:2: ALWAYS 0x555556b4a210 <e42> {c5ae}
    1:2:2: IF 0x555556b4e000 <e110> {c6ah}
    1:2:2:1: LOGAND 0x555556b4a420 <e162#> {c6an} @dt=0x555556b08680@(G/w1)
    1:2:2:1:1: EQ 0x555556b4a2c0 <e158#> {c6an} @dt=0x555556b08680@(G/w1)
    1:2:2:1:1:1: CONST 0x555556b4c500 <e140#> {c6an} @dt=0x555556b08680@(G/w1)  1'h0
    1:2:2:1:1:2: VARREF 0x555556b027e0 <e141#> {c6an} @dt=0x555556b08680@(G/w1)  a__en [RV] <- VAR 0x555556b00600 <e130#> {c2aj} @dt=0x555556b08680@(G/w1)  a__en MODULETEMP
    1:2:2:1:2: EQCASE 0x555556b4a370 <e159#> {c6an} @dt=0x555556b08680@(G/w1)
    1:2:2:1:2:1: CONST 0x555556b4c640 <e154#> {c6an} @dt=0x555556b08680@(G/w1)  1'h0
    1:2:2:1:2:2: VARREF 0x555556b02480 <e155#> {c6al} @dt=0x555556b08680@(G/w1)  a [RV] <- VAR 0x555556b00300 <e91> {c2aj} @dt=0x555556b08680@(G/w1)  a [VSTATIC]  WIRE
    1:2:2:2: STOP 0x555556b4a160 <e38> {c7ak}
    1:2: VAR 0x555556b00600 <e130#> {c2aj} @dt=0x555556b08680@(G/w1)  a__en MODULETEMP
    3: TYPETABLE 0x555556afe000 <e2> {a0aa}
                   logic  -> BASICDTYPE 0x555556b08680 <e28> {c6ar} @dt=this@(G/w1)  logic [GENERIC] kwd=logic
                detailed  ->  BASICDTYPE 0x555556b08680 <e28> {c6ar} @dt=this@(G/w1)  logic [GENERIC] kwd=logic
    3:1: BASICDTYPE 0x555556b08680 <e28> {c6ar} @dt=this@(G/w1)  logic [GENERIC] kwd=logic
    3:1: BASICDTYPE 0x555556b081a0 <e69> {c1aw} @dt=this@(w1)  LOGIC_IMPLICIT kwd=LOGIC_IMPLICIT
    3:1: BASICDTYPE 0x555556b08340 <e77> {c2aj} @dt=this@(w1)  logic kwd=logic
    3: CONSTPOOL 0x555556b00000 <e6> {a0aa}
    3:1: MODULE 0x555556b02000 <e4> {a0aa}  @CONST-POOL@  L0 [NONE]
    3:1:2: SCOPE 0x555556af80f0 <e5> {a0aa}  @CONST-POOL@ [abovep=0] [cellp=0] [modp=0x555556b02000]
```
The problem is that `a__en` is not initialized, so it equals `0` I think.
So I would say that we need to mark `clk` as tristate (I think it should be done according to this comment: https://github.com/antmicro/verilator-1/blob/937e893b6d689a641d2d0be91cfec1f40ff05dca/src/V3Tristate.cpp#L34-L35), then it would be handled correctly by visitAssign (https://github.com/antmicro/verilator-1/blob/937e893b6d689a641d2d0be91cfec1f40ff05dca/src/V3Tristate.cpp#L914-L916).
But I don't know why tristates should be propagated from LHS to RHS. Maybe it would be better to assign `'1` to `__en` variable if its original variable is on the LHS of assignment, when the RHS is not tristate?

wsnyder, 2022-09-07:
There's a performance cost to propagating the tristate.

I wonder instead should we simply assume any comparison with Z on a non-tristate net is false?

RRozak, 2022-09-08:
The problem is that some nodes marked as non-tristate (by non-tristate I understand that they don't have user1p) are actually tristates. That's why in this PR https://github.com/verilator/verilator/pull/3604/ I added a function that returns enable expression if user1p() is not set. Now I think that this way maybe is not the best one. Maybe it would be better to make sure that each node that may contain `z` value has user1p()?

wsnyder, 2022-09-13:
>Maybe it would be better to make sure that each node that may contain z value has user1p()?

I think that might work well, as that is basically suggesting the user does want 3-state intent.  You might still need to deal with the variable being compared to (might still not have a user1 in the visitCaseEq/visitEqNeqWild visitor, so that visitor code would need to know that it is false (depending on bits that are Z and type of compare.)

wsnyder, 2022-09-13:
BTW please make sure your test has ==, !==, ===, !===.