verilator/verilator #3591: Debug verilator itself and how to configure when building verilator from src

MJChku, 2022-09-02:
I want to debug verilator (its passes and graphs), so I want to use vscode/gdb, and supply -g option when building verilator from src. Is there an easy way to do that through ./configure (I didn't find any). So what can I do ? essentially I want to set breakpoint at source files and debugger will stop there.

wsnyder, 2022-09-02:
A debug executable is built by default (verilator_dbg). Just use "verilator --gdb {flags}" and it will get run.

wsnyder, 2022-09-02:
Or --gdbrun to save having to run manually in gdb.

closed 2022-09-02 by wsnyder

gezalore, 2022-09-02:
Additionally might want to change `-Og` to `-O0` in `configure.ac` for a better experience. See also: https://github.com/verilator/verilator/blob/master/docs/internals.rst#debugging

Otherwise I usually just go dirty and `./configure CXX="g++ -<whatever_flag_is_needed>"` if I just want to stick some extra options on.

MJChku, 2022-09-02:
I want to combine c++ debugger in vscode, so to set breakpoint while browsing through verilator src files. But yeah, it's there already, run  "verilator_bin_dbg" instead of "verilator_bin" (configure in vscode configuration).