verilator/verilator #3602: Assignment to string in a struct produces confusing internal errors

kiryk, 2022-09-06:
Assigning to string fields in structs results in an internal error during V3Number pass.
I also find the error message rather confusing.

```
%Error: Internal Error: t/t_struct_string.v:22:14: ../V3Number.cpp:1747: Number operation called with non-logic (double or string) argument: '"content""
   22 |       st.a = "content";
      |              ^~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

It seems that struct accesses are transformed into SELs at some point, which are number operations, although strings are not allowed in packed structs by IEEE at all, so probably this should be detected during earlier passes.

Interestingly enough, the error doesn't occur when the string is the only field of a struct.

The error can be reproduced with a pair of files in `test_regress` format that can be downloaded [here](https://github.com/verilator/verilator/files/9496400/t_struct_string.tar.gz).

I use Debian Linux with the newest Verilator as for this moment (e94cdcf29c22fac332ca42fc711d0e186e7d8369).

wsnyder, 2022-09-08:
This is due to #1541.

Ideally it should give a much cleaner UNSUPPORTED error, so will leave this open with that suggestion.  (Well, really ideally, #1541 would be fixed to support it.)

kiryk, 2022-09-08:
Okay then. I'll try to add support for unpacked structs.

closed 2022-12-16 by wsnyder