verilator/verilator #3603: Namespacing of generated code

gezalore, 2022-09-06:
I was just trying to build the same design with 2 different options, and was planning to compare output variables for equivalence (for DFG testing), so I went:

```
verilator --cc -Mdir obj_ref -fno-dfg --prefix Vref t_dfg_peephole.v
verilator --cc -Mdir obj_opt          --prefix Vopt t_dfg_peephole.v
```
to build the 2 versions. The output directories then contain:
```
[gl@flatbox test_regress] % ll obj_vlt/t_dfg_peephole/obj_opt
total 580K
 20K -rw-rw-r-- 1 gl gl  20K Sep  6 16:57 Vopt__ALL.a
4.0K -rw-rw-r-- 1 gl gl  445 Sep  6 16:57 Vopt__ALL.cpp
4.0K -rw-rw-r-- 1 gl gl  730 Sep  6 16:57 Vopt__ALL.d
 20K -rw-rw-r-- 1 gl gl  18K Sep  6 16:57 Vopt__ALL.o
4.0K -rw-rw-r-- 1 gl gl 1.7K Sep  6 16:57 Vopt_classes.mk
4.0K -rw-rw-r-- 1 gl gl 2.6K Sep  6 16:57 Vopt.cpp
4.0K -rw-rw-r-- 1 gl gl 2.7K Sep  6 16:57 Vopt.h
4.0K -rw-rw-r-- 1 gl gl 1.5K Sep  6 16:57 Vopt.mk
 72K -rw-rw-r-- 1 gl gl  69K Sep  6 16:57 Vopt__stats.txt
4.0K -rw-rw-r-- 1 gl gl  749 Sep  6 16:57 Vopt__Syms.cpp
4.0K -rw-rw-r-- 1 gl gl  904 Sep  6 16:57 Vopt__Syms.h
4.0K -rw-rw-r-- 1 gl gl 1008 Sep  6 16:57 Vopt__ver.d
4.0K -rw-rw-r-- 1 gl gl 2.7K Sep  6 16:57 Vopt__verFiles.dat
324K -rw-rw-r-- 1 gl gl 323K Sep  6 16:57 Vopt.xml
4.0K -rw-rw-r-- 1 gl gl 1023 Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_h83248e7a__0.cpp
4.0K -rw-rw-r-- 1 gl gl  910 Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_h83248e7a__0__Slow.cpp
 44K -rw-rw-r-- 1 gl gl  43K Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_hb2c8af09__0.cpp
 44K -rw-rw-r-- 1 gl gl  44K Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_hb2c8af09__0__Slow.cpp
4.0K -rw-rw-r-- 1 gl gl 1.1K Sep  6 16:57 Vt_dfg_peephole___024root.h
4.0K -rw-rw-r-- 1 gl gl  707 Sep  6 16:57 Vt_dfg_peephole___024root__Slow.cpp
[gl@flatbox test_regress] % ll obj_vlt/t_dfg_peephole/obj_ref
total 1016K
 16K -rw-rw-r-- 1 gl gl  15K Sep  6 16:57 Vref__ALL.a
4.0K -rw-rw-r-- 1 gl gl  445 Sep  6 16:57 Vref__ALL.cpp
4.0K -rw-rw-r-- 1 gl gl  730 Sep  6 16:57 Vref__ALL.d
 16K -rw-rw-r-- 1 gl gl  13K Sep  6 16:57 Vref__ALL.o
4.0K -rw-rw-r-- 1 gl gl 1.7K Sep  6 16:57 Vref_classes.mk
4.0K -rw-rw-r-- 1 gl gl 2.6K Sep  6 16:57 Vref.cpp
4.0K -rw-rw-r-- 1 gl gl 2.7K Sep  6 16:57 Vref.h
4.0K -rw-rw-r-- 1 gl gl 1.5K Sep  6 16:57 Vref.mk
 44K -rw-rw-r-- 1 gl gl  44K Sep  6 16:57 Vref__stats.txt
4.0K -rw-rw-r-- 1 gl gl  749 Sep  6 16:57 Vref__Syms.cpp
4.0K -rw-rw-r-- 1 gl gl  904 Sep  6 16:57 Vref__Syms.h
4.0K -rw-rw-r-- 1 gl gl 1008 Sep  6 16:57 Vref__ver.d
4.0K -rw-rw-r-- 1 gl gl 2.7K Sep  6 16:57 Vref__verFiles.dat
688K -rw-rw-r-- 1 gl gl 688K Sep  6 16:57 Vref.xml
4.0K -rw-rw-r-- 1 gl gl 1023 Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_h426b2197__0.cpp
4.0K -rw-rw-r-- 1 gl gl  910 Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_h426b2197__0__Slow.cpp
 96K -rw-rw-r-- 1 gl gl  96K Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_hb2c8af09__0.cpp
100K -rw-rw-r-- 1 gl gl  97K Sep  6 16:57 Vt_dfg_peephole___024root__DepSet_hb2c8af09__0__Slow.cpp
4.0K -rw-rw-r-- 1 gl gl 1.1K Sep  6 16:57 Vt_dfg_peephole___024root.h
4.0K -rw-rw-r-- 1 gl gl  707 Sep  6 16:57 Vt_dfg_peephole___024root__Slow.cpp
```

Hashes differ but the corresponding Vt_dfg_peephole___024root* files define functions with identical signatures, so I can't link these two models into the same executable. I can bodge it renaming the module with `--pipe-filter`, but is there a better way? I can't seem to find one.

Questions:
- Is this expected?
- Should --prefix apply to all symbols defined? In the case where `--prefix` is defaulted to the name of the top-level, it kind of does, but that seems to be only an accident at this point. Should we just `s/toplevel/prefix/` everywhere so the we would get for example "Vref___024root*" in the above? Alternatively could wrap everything in a user specified `namespace`, though one still ends up with file name clashes in that case.

wsnyder, 2022-09-08:
The prefix should apply -- I think we missed that when we added DepSet changes.

gezalore, 2022-09-12:
So there is also `--mod-prefix`, and reading the help it's non-obvious what is expected.

There are at least these things we need to consider:
1. Top level class name (as in, the VeriltedModel subclass) and makefile: use `--prefix` (this is documented under `--prefix`).
2. Top level class .h and .cpp file: use `--prefix`
3. Other class, function, and in general any global symbol names (relevant for linking): What to use here `--mod-prefix` or `--prefix`, or a mix?
4. Other .h and .cpp files: Use whatever we use for point 3 I guess

In general I'm not sure what the use-case is for having a separate `--mod-prefix`, maybe deprecate it/make it ignored?

gezalore, 2022-09-12:
Turns out this is a quirk of option parsing combined with driver.pl, where driver.pl always adds an extra initial `--prefix` option, and `--mod-prefix` used to default to the first `--prefix`. Fixed by defaulting to the last `--prefix` if `--mod-prefix` not given.

commit 08b6bddd (2022-09-12): Update default --mod-prefix when --prefix is repeated

Fixes #3603
closed 2022-09-12 by gezalore