verilator/verilator #3611: Unexpected IDENTIFIER process::state in OVM-2.1.2

benjaminmordaunt, 2022-09-11:
(NOTE: I know there's still some work to be done to fully enable the likes of OVM/UVM support, but I want to help narrow down the work that needs doing)

Verilator 4.227 devel rev v4.226-17-g45d622e4
macOS arm64

When trying to run the "hello_world" example from the OVM 2.1.2 kit, I receive the following syntax error from `ovm_component.sv:653:18`:

```
%Error: /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_component.sv:653:18: syntax error, unexpected IDENTIFIER
  653 |   process::state ps;
      |                  ^~
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_registry.svh:283:1: ... note: In file included from ovm_registry.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_factory.sv:1058:1: ... note: In file included from ovm_factory.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_transaction.sv:333:1: ... note: In file included from ovm_transaction.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_report_handler.svh:610:1: ... note: In file included from ovm_report_handler.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_report_server.svh:407:1: ... note: In file included from ovm_report_server.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_barrier.svh:208:1: ... note: In file included from ovm_barrier.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_event.svh:358:10: ... note: In file included from ovm_event.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_event_callback.svh:87:1: ... note: In file included from ovm_event_callback.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_recorder.svh:208:1: ... note: In file included from ovm_recorder.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_packer.sv:491:1: ... note: In file included from ovm_packer.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_printer.sv:1086:1: ... note: In file included from ovm_printer.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_pool.svh:337:59: ... note: In file included from ovm_pool.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_object.sv:1320:1: ... note: In file included from ovm_object.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/ovm_misc.sv:517:1: ... note: In file included from ovm_misc.sv
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/base/base.svh:25:68: ... note: In file included from base.svh
        /Users/benjaminmordaunt/Downloads/ovm-2.1.2/src/ovm_pkg.sv:24:16: ... note: In file included from ovm_pkg.sv
```

Why would Verilator see this as a syntax error? Is it something to do with the use of namespacing for `process::state`?

wsnyder, 2022-09-11:
Are you really using OVM vs UVM?  OVM was basically replaced by UVM 10 years ago, so I'm surprised is still of any interest?

As to the issues many overlap UVM where the focus is more common; it's better to track this in #1538 where I see you already commented.

benjaminmordaunt, 2022-09-11:
:) Some legacy stuff, including still-licensed IP cores use OVM. I acknowledge many of the issues overlap with UVM enablement.
I guess it's up to you then if you want to close this in favour of tracking only against UVM. 

Personally, I would like to see separate tracking issues for many common verification libraries so that we can see when Verilator reaches each support milestone, which issues are outstanding against each library etc.

wsnyder, 2022-09-11:
>I would like to see separate tracking issues for many common verification libraries 

Yes, agreed. That's reasonable, I just wanted to make sure it wasn't a case of UVM didn't work so you tried OVM.

Note OVM might need a slight edit to correspond what we did to UVM to get around #2412.  That won't get you much further until o#1538 makes more progress.

closed 2025-02-25 by wsnyder