verilator/verilator #3613: Compilation error - foreach on map inside a task

nndurj, 2022-09-11:
### Getting this error
```
 make
%Error: Internal Error: tb_top.sv:8:17: ../V3Task.cpp:127: No scope for var
    8 |     foreach(map[key]) $display("key: %0d", key);
      |                 ^~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make: *** [Makefile:65: build/tb_top.mk] Error 1

```
### Source
```systemverilog
module tb_top (
  input v_clk
);

  int map[int];

  task disp();
    foreach(map[key]) $display("key: %0d", key);
  endtask

  initial begin
    for (int i=0; i<10; i++) map[i] = i+10;
    disp();
  end

endmodule

```

### version 
```
verilator --version
Verilator 4.227 devel rev v4.226-17-g45d622e4
```

wsnyder, 2022-09-11:
If used outside a function it seems to work.  Perhaps you'd be willing to dig into why it isn't working and propose a pull request?

jensweb, 2022-09-29:
Observing similiar issue with increment in array index. It does work outside of functions as well. 

```
module test(clk);
   input clk;      
   function bit[1:0][1:0] func_nok();
     int i;
     i = 0;
     // does not work
     func_nok[i++] = 0;
     // does work
     //func_nok[i] = 0; i++;
     func_nok[i] = 0;
   endfunction
endmodule

```
verilator --version
Verilator 4.227 devel rev v4.226-72-gf25a9a4c8