verilator/verilator #3618: Syntax for a Public SystemVerilog Enum Using `verilator_config

bobnewgard, 2022-09-12:
I am having difficulty making an SV enum public using the `verilator_config mechanism.

The enum is defined in a package:

```
package bbus_pkg;
    typedef enum [1:0]
    {
        BBUS_CS_NIL,
        BBUS_CS_WR,
        BBUS_CS_RD,
        BBUS_CS_ERR
    }
    t_bbus_cs;
endpackage
```

What is the correct `verilator_config syntax for this?  It's not successful with

```
`verilator_config
public -module "bbus_pkg" -var "t_bbus_cs"
```

It's also not successful with

```
`verilator_config
public -module -var "t_bbus_cs"
```

What am I doing wrong?

FYI, I am able to make it public using /* verilator public */, but I prefer to go the `verilator_config route.

wsnyder, 2022-09-12:
It looks like we don't have a test like t_enum_public that uses the VLT syntax (as we do for e.g. t_trace_public_func.vlt), so probably this has never been supported.  I'd presume the syntax would be:

      public -enum "t_bbus_cs"

Might you be willing to make such a test, then work on a pull request to fix it?  We'd need an update to docs/guide files, parser in verilog.l and verilog.y to recognize --enum, then V3Config to make the list of enums to make public.

bobnewgard, 2022-09-13:
Added a pull request #3619 for a test.

bobnewgard, 2022-09-13:
I don't have time at the moment to attempt the implementation, but thanks to your  feature implementation roadmap, I may be able to do it over time.