verilator/verilator #3624: Incorrect byte concatenation

Thomaswsup, 2022-09-14:
_Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)_
`gy.v`
```verilog
// We get 8bit data in every clock
// Output back as 16bit data, updated every second clock

module gy (input         igy_clk,
           input [7:0]   igy_data_7_0,
           output [15:0] gy_outdata_15_0);

  //-------- Variables ------------------------------------------------------

  reg        ngy_inctr;
  reg [7:0]  ngy_inreg_7_0;
  reg [15:0] ngy_dinreg_15_0;

  reg        gy_inctr = 0;
  reg [7:0]  gy_inreg_7_0 = 0;
  reg [15:0] gy_dinreg_15_0 = 0;

  //-------------------------------------------------------------------------

  always @ * begin

    ngy_inctr = !gy_inctr;

    ngy_inreg_7_0 = igy_data_7_0;

    if (gy_inctr == 0)
      ngy_dinreg_15_0 = {gy_inreg_7_0,
                         igy_data_7_0};
    else
      ngy_dinreg_15_0 = gy_dinreg_15_0;

  end // always @ *

  always @ (posedge igy_clk) begin
    if (0) begin
      $write("\n");
      $write("igy_data_7_0        = %02x\n",  igy_data_7_0);
      $write("gy_inctr        = %01x",     gy_inctr);
      $write("   ngy_inctr        =     %01x\n", ngy_inctr);
      $write("gy_inreg_7_0    =   %02x",   gy_inreg_7_0);
      $write("   ngy_inreg_7_0    =   %02x\n",   ngy_inreg_7_0);
      $write("gy_dinreg_15_0  = %04x",     gy_dinreg_15_0);
      $write("   ngy_dinreg_15_0  = %04x\n",     ngy_dinreg_15_0);
      $write("\n");
    end
  end

  always @ (posedge igy_clk) begin
    gy_inctr        <= ngy_inctr;
    gy_inreg_7_0    <= ngy_inreg_7_0;
    gy_dinreg_15_0  <= ngy_dinreg_15_0;
  end

  assign gy_outdata_15_0 = gy_dinreg_15_0;

endmodule

```
`gy.cpp`
```c++
#include <verilated.h>
#include <iostream>
#include "Vgy.h"

Vgy* top;

int main(int argc, char **argv) {
  Verilated::commandArgs(argc, argv);
  unsigned int cnt = 0;

  top = new Vgy;

  while (!Verilated::gotFinish()) {
    top->igy_clk = 0;
    top->eval();

    top->igy_clk = 1;
    top->igy_data_7_0 = cnt;
    top->eval();
    printf("outdata = 0x%04x\n", top->gy_outdata_15_0);

    cnt++;
    if(cnt > 15){
      printf("DONE\n");
      break;
    }
  }
  top->final();
  delete top;
}
```
_Build command_ 
verilator -Wall +1364-2001ext+v --cc --exe --build gy.cpp gy.v

_What 'verilator --version' are you using?  Did you try it with the git master version?_
4.227 devel rev v4.226-26-g25644844

_What OS and distribution are you using?_
Debian Bullseye

_Description_
-------------------
Expected output:

$ obj_dir/Vgy
outdata = 0x0000
outdata = 0x0000
outdata = 0x0102
outdata = 0x0102
outdata = 0x0304
outdata = 0x0304
outdata = 0x0506
outdata = 0x0506
outdata = 0x0708
outdata = 0x0708
<...>

Actual output:

$ obj_dir/Vgy
outdata = 0x0000
outdata = 0x0000
outdata = 0x0101
outdata = 0x0101
outdata = 0x0303
outdata = 0x0303
outdata = 0x0505
outdata = 0x0505
outdata = 0x0707
outdata = 0x0707
<...>

Looks like the data actually latched into gy_dinreg_15_0 is corrupted.

When running gy.v via iverilog, outdata looks as expected.

gezalore, 2022-09-14:
>    top->igy_clk = 1;
>    top->igy_data_7_0 = cnt;
>    top->eval();

It is generally not safe to change both clocks and data on the same `eval` call, please try:
```
    top->igy_data_7_0 = cnt;
    top->eval();
    top->igy_clk = 1;
    top->eval();
```

That is: change all data inputs, call `eval`, then change clocks, call `eval`

Or try the develop-v5 branch where this has been improved and should work as you might expect.

Thomaswsup, 2022-09-14:
Thanks for the quick reply!
Adding an extra eval() makes things work as expected.
As you suggested, develop-v5 also makes everything work as-is.
Thanks again.

closed 2022-09-14 by Thomaswsup