verilator/verilator #3625: Consider option for running simulation

gezalore, 2022-09-14:
Now that v5 has timing support and hand-written C++ is no longer required, I wonder if it would be helpful for new/simple users to have an option like `--run`, which verilates, compiles, and runs the simulation. Currently you have to type:
```
verilator --cc --timing --main --exe --build foo.v
./obj_dir/Vfoo
```
Instead could just be
```
verilator --run foo.v
```

I don't have strong feelings about it and I am unlikely to use this myself, but might help lower the entry barrier?

wsnyder, 2022-09-14:
I don't want to support --run, because I think it combines two things that should be separated in the user's mind to avoid rebuild penalties.  Also more trivially bin/verilator needs to handle runtime flags and that's a  whole collision to avoid.

Maybe we should change --main to imply --cc, as it is an output mode of sorts.

Perhaps a new option to combine "--timing --main --exe --build"?   It's too bad --exe already is taken, as that's what it is creating.

wsnyder, 2022-09-14:
Maybe --bin?

gezalore, 2022-09-14:
Fair enough regarding the compile/run separation (that is until the hot-swappable JIT backend arrives ;))

> Maybe we should change --main to imply --cc, as it is an output mode of sorts.

Probably should do that as I guess --main --sc would bomb anyway?

gezalore, 2022-09-14:
> Maybe --bin?

Could be, or --sim / --sim-exe / --sim-bin ?

wsnyder, 2022-09-15:
I pushed a commit to have --main imply -cc. Also I realized that --main was still internal only, so pushed documenting it.

wsnyder, 2022-09-15:
I realized that --bin was taken, it's a rare option that has a parameter so we could still take it (since an error will happen if someone uses the old form).  It might make sense to rename --bin as --exe-name regardless.

wsnyder, 2022-09-26:
--bin was renamed, thinking of --binary for this.

closed 2022-09-28 by wsnyder

gezalore, 2022-09-28:
Shouldn't this also add --timing? (unless --no-timing was given explicitly?)