verilator/verilator #3631: warning not disabled despite /* verilator lint_off MULTIDRIVEN */ and /* verilator lint_on MULTIDRIVEN */ around the code

gkamendje, 2022-09-19:
Verilator seems to ignore /* verilator lint_off MULTIDRIVEN */ and /* verilator lint_on MULTIDRIVEN */ that I place around the code to disable the `Warning-MULTIDRIVEN` warning. However, the options `-Wno-MULTIDRIVEN` on the command line does suppress the warning. But this is not what I want. 
Using the following code 
```
module test 
#(parameter W_CNT=11) 
(
    input  logic i_rst_n      , // active low asynchronous reset
    input  logic i_clk        , // clock
    output logic [W_CNT-1:0] o_q
);
//------------------------------------------
//VARS
//------------------------------------------
genvar i;
always_ff @(posedge i_clk or negedge i_rst_n) begin
    if ( i_rst_n== 1'b0) begin
        o_q[0] <= 1'b1;  
    end else begin
/* verilator lint_off MULTIDRIVEN */
            o_q[0] <= ! o_q[0]; 
/* verilator lint_on MULTIDRIVEN */
    end
end
  generate
      for (i=1; i <= W_CNT-1; i++) begin : async_o
          always_ff @( negedge o_q[i-1] or negedge i_rst_n) begin
              if ( i_rst_n== 1'b0) begin
                  o_q[i] <= 1'b0;  
              end else begin 
/* verilator lint_off MULTIDRIVEN */
                  o_q[i] <= ! o_q[i]; 
/* verilator lint_on MULTIDRIVEN */
              end
          end
      end : async_o
  endgenerate
endmodule
```
And running it trough verilator
`/cad/verilator/4.226/bin/verilator --lint-only test.sv --top-module test`
I do get the following error message.
```
%Warning-MULTIDRIVEN: test.sv:15:30: Signal has multiple driving blocks with different clocking: 'o_q'
                      test.sv:37:19: ... Location of first driving block
   37 |                   o_q[i] <= ! o_q[i]; 
      |                   ^~~
                      test.sv:26:13: ... Location of other driving block
   26 |             o_q[0] <= ! o_q[0]; 
      |             ^~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=4.226
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Warning-UNOPT: test.sv:15:30: Signal unoptimizable: Feedback to public clock or circular logic: 'o_q'
   15 |     output logic [W_CNT-1:0] o_q
      |                              ^~~
%Error: Exiting due to 2 warning(s)
```
I have tried putting the lint_off/on in different location in the code but the result is the same. 

BTW: I do not understand why I am getting the `Warning-UNOPT` warning either. 

What 'verilator --version' are you using?  
Verilator 4.226 2022-08-31 rev v4.226
What OS and distribution are you using?
Centos7

gezalore, 2022-09-19:
The UNOPT is due to `o_q[_] <= o_q[_]` and `o_q` being used in a sensitivity list, and because Verilator does this analysis (and it's internal scheduling by signal name. Consider using the `split_var` directive to make Verilator split the signal for you: https://verilator.org/guide/latest/extensions.html?highlight=split_var#cmdoption-verilator-32-split_var

Not sure about the MULTIDRIVEN, putting the lint_off around the signal definition should remove it.

wsnyder, 2022-09-19:
Right, note the warning itself is on "test.sv:15" which is where o_q is declared so that is what needs the disable.

Closing as think answered, but feel free to follow up with more questions.

closed 2022-09-19 by wsnyder

wsnyder, 2022-09-19:
You will need to make a new variable you split_var, then "assign o_q =" from that.

gkamendje, 2022-09-19:
Ok thanks