verilator/verilator #3633: Reduce AstNode memory consumption by storing operands before `this`

gezalore, 2022-09-21:
Now that we have metadata on the arity of AstNodes, we could tweak the representation so that instead of

```
this -> +-----------
        | vtablep
        +-----------
        | m_nextp
        +-----------
        | m_backp
        +-----------
        | m_op1p
        +-----------
        | m_op2p
        +-----------
        | m_op3p
        +-----------
        | m_op4p
        +-----------
        | m_iterpp
        +-----------
```
we lay out memory as 

```
        +-----------
        | m_op4p
        +-----------
        | m_op3p
        +-----------
        | m_op2p
        +-----------
        | m_op1p
this -> +-----------
        | vtablep
        +-----------
        | m_nextp
        +-----------
        | m_backp
        +-----------
        | m_iterpp
        +-----------
```

The benefit is that we can then not allocate storage to the operands we know we never use. This can be implemented with the proper customisation of new and delete.

Currently `sizeof(AstNode)` is 176. I estimate on average we only need 2.2 operands per AstNode, which would save about 14.5 bytes, so we are looking at a reduction of somewhere between 5-10%?

@wsnyder any objections?

gezalore, 2022-09-21:
Note: there a lot of places in V3Ast.cpp that needs fixing up, some of them performance critical that needs special treatment, some of it can be abstracted over by checking the node arity.

wsnyder, 2022-09-21:
I have made similar specialization experiments overt the years, and they so far have been performance losses. The problem is once not all nodes are the same you have a branch that is impossible to predict, and that pipeline cost more than trumps the other gains.

Feel free to experiment.  Also if the gain is only modest I would avoid such complexity.

wsnyder, 2022-10-21:
Closing as this isn't something we have to do - but still welcome data on if it helps.

closed 2022-10-21 by wsnyder