verilator/verilator #3634: Comment lines that start with verilator trigger strange error message and kill the run

chick, 2022-09-22:
# Verilator comment pattern breaks verilation

## Version
```
$ verilator --version
Verilator 4.224 2022-06-19 rev UNKNOWN.REV
```

## Example file 
The HEREIS doc generates an example .sv file
Note the comment line's structure
```
cat > BlackBoxAdderWrapper.sv <<-EOF
module BlackBoxAdderWrapper(
  input        clock,
               reset,
  input  [7:0] io_a,
               io_b,
  output [7:0] io_q);

  BlackBoxAdder m (	// VerilatorExample.scala:33:19
    .a (io_a),
    .b (io_b),
    .q (io_q)
  );
endmodule

// -------
module BlackBoxAdder(a, b, q);
  input [7:0] a, b;
  output [7:0] q;
  assign q = a + b;
endmodule
EOF
```

## Run verilator with this command
The HEREIS document creates a bash script to run verilator with a typical flag configuration
```
cat > run_verilator.sh <<-EOF
verilator \
--cc \
--exe \
BlackBoxAdderWrapper-harness.cpp \
--assert \
--coverage-user \
-Wno-fatal \
-Wno-WIDTH \
-Wno-STMTDLY \
--top-module \
BlackBoxAdderWrapper \
+define+TOP_TYPE=VBlackBoxAdderWrapper \
-CFLAGS \
    "-O1 \
    -DVL_USER_STOP \
    -DVL_USER_FATAL \
    -DVL_USER_FINISH \
    -DTOP_TYPE=VBlackBoxAdderWrapper \
    -include \
    VBlackBoxAdderWrapper.h \
    -fPIC \
    -shared \
    -I'/Library/Java/JavaVirtualMachines/jdk-11.0.13.jdk/Contents/Home/include' \
    -I'/Library/Java/JavaVirtualMachines/jdk-11.0.13.jdk/Contents/Home/include/darwin' \
    -fvisibility=hidden" \
-Mdir \
./verilated \
-LDFLAGS \
    "-shared \
    -dynamiclib \
    -fvisibility=hidden" \
BlackBoxAdderWrapper.sv
EOF
```

## Fails to complete
run the bash script `sh run_verilator.sh` and the
output of the operation for me reports the error
``` 
%Error: BlackBoxAdderWrapper.sv:8:21: Unknown verilator comment: '/*verilator Example.scala:33:19*/'
    8 |   BlackBoxAdder m ( /*verilator Example.scala:33:19*/ 
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
```

## Incontrast
Remove the comment from the line 
``` 
  BlackBoxAdder m (	// VerilatorExample.scala:33:19
```
to 
```
  BlackBoxAdder m (	// OtherCommentExample.scala:33:19
```
And rerun the `run_verilator.sh` runs successfully.
It seems like a comment that begins with verilator kicks off something.

wsnyder, 2022-09-22:
As described in the documentation, // verilator is a metacomment used to disable warnings and other things, so this error is intentional to detect typos.

closed 2022-09-22 by wsnyder

seldridge, 2022-09-22:
Though, this is indicating that `// Verilatorfoo` is interpreted equivalently as the metacomment `// verilator foo`. Is the splitting and case insensitivity the intended behavior?

wsnyder, 2022-09-22:
Yes, this is intended.