verilator/verilator #3640: Support output report to file when doing lint

hwzp, 2022-09-26:
When I use --lint-only to do lint, the results are all printed on the screen, and I can't output the lint results to a file. I can't see all the warning or error messages on the screen.

wsnyder, 2022-09-26:
I presume you are new to Linux. You can redirect the output of any program to a file using an appropriate pipe.  See for example https://askubuntu.com/questions/420981/how-do-i-save-terminal-output-to-a-file

Alternatively all good editors have a way to run a command within the editor, this makes it easy to move around the output. You'll need to search for how to do that for your editor.  (Or if you aren't using an editor that supports it, it will do your long term productivity good to switch.)

closed 2022-09-26 by wsnyder

hwzp, 2022-09-27:
Thank you!I did a new try and now the problem is solved.