verilator/verilator #3641: String error: no known conversion from 'VlWide<5>' to 'initializer_list<char>' for 1st argument

dpretet, 2022-09-26:
Hello,

I wrote a basic SystemVerilog class to make my prints prettier and well formatted. Until now I always used it with Icarus but now I'm using more and more Verilator, I try to use it with it too. However, I experience an error I don't understand:

```
...
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/string:822:5: note: candidate constructor not viable: no known conversion from 'VlWide<5>' to 'const std::string &' for 1st argument
    basic_string(const basic_string& __str);
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/string:827:5: note: candidate constructor not viable: no known conversion from 'VlWide<5>' to 'std::string &&' for 1st argument
    basic_string(basic_string&& __str)
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/string:840:5: note: candidate constructor template not viable: no known conversion from 'VlWide<5>' to 'const char *' for 1st argument
    basic_string(const _CharT* __s) : __r_(__default_init_tag(), __default_init_tag()) {
    ^
...

```

I'm using Verilator 4.226 on macOS Monterey 12.4. You can reproduce this issue to have more messages by cloning 'develop-verilator' branch and go to example folder and execute `./runme.sh --simulator verilator`. The repository is here: 

[SVLogger](https://github.com/dpretet/svlogger/tree/develop-verilator)

The code consist of a simple class in this [file](https://github.com/dpretet/svlogger/blob/develop-verilator/svlogger.sv)

Could you help me please to understand the problem and if I made a mistake somehow with Verilator.

Damien

wsnyder, 2022-09-27:
There is no operator<< defined for VlWide at present, there is however a VL_TO_STRING for it.  Perhaps you could contribute a pull of code to have operator<< call VL_TO_STRING, and add appropriate tests?

Ideally add similar for the other data structures, VlQueue, VlAssocArray, VlUnpacked.

dpretet, 2022-09-27:
I'm not a C++ developer but maybe I could do that if you point the sources and the tests I need to modify, that could be fun to do :)

wsnyder, 2022-09-27:
There's information on writing tests etc in docs/internal.txt, I think you could use $c in verilog to e.g. `$c("cout<<", q << ";");`. 

The classes needing the operator are in include/verilated_types.h.

wsnyder, 2022-10-18:
This passes on the current master, can you confirm this is ok?

dpretet, 2022-10-18:
Hello,

I tried the master (commit 51e3f15dce), compiled it on Ubuntu 2020.04 up-to-date, and everything is fine :) Thank you very much!

closed 2022-10-18 by dpretet

dpretet, 2022-10-18:
Quick question: When would it b delivered in a new release?

wsnyder, 2022-10-18:
Likely next release is end of month (ish).