verilator/verilator #3646: %Error: ... Cannot add models to a trace file if 'dump' has already been called

bobnewgard, 2022-09-28:
[vltr_trace_err.tar.gz](https://github.com/verilator/verilator/files/9660576/vltr_trace_err.tar.gz)

Error emitted at the trace() function:

```
#include <tb.h>
int sc_main(int argc, char **argv)
{
    Verilated::commandArgs(argc, argv);
    Verilated::debug(0);
    Verilated::traceEverOn(true);
    std::unique_ptr<tb::tb> tb  = (std::unique_ptr<tb::tb>)(new tb::tb("tb"));
    VerilatedVcdSc *tfp = new VerilatedVcdSc;
    cerr << "[DBG] 1" << endl << flush;
    sc_start(1, SC_NS);
    cerr << "[DBG] 2" << endl << flush;
    tb->i_tb->trace(tfp, 99);
    cerr << "[DBG] 3" << endl << flush;
    return 0;
}
```

Error found in v4.226, but doesn't occur in v4.224.

The TGZ contains a minimized set of test case files, log files, a Readme file and a test.bash test driver script.  The Readme shows the interpolation results and how to run ./test.bash for various release tags.  Log file v4.226.log shows the error, and fails to print '[DBG] 3' while v4.224.log shows the expected behavior.

I don't doubt that I'm doing something wrong, but what is it?

wsnyder, 2022-09-28:
You need to call ->trace before you ever call sc_start.

closed 2022-09-28 by wsnyder

bobnewgard, 2022-09-28:
Swapping the calls to sc_start() and trace() fails on trace() with

```
%Error: Vex1_group6.cpp:119: Vex1_group6::trace() is called before sc_core::sc_start(). Run sc_core::sc_start(sc_core::SC_ZERO_TIME) before trace() to complete elaboration.
```

But after implementing the suggested replacement of ```sc_start(1, SC_NS)``` with ```sc_start(SC_ZERO_TIME)```, the test was successful.

Note that the in the make_tracing_sc example, sc_start(1, SC_NS) is called before top->trace().

bobnewgard, 2022-09-28:
```sc_start(SC_ZERO_TIME)``` also fixed the issue in my working project.