verilator/verilator #3647: I made a funny thing

quantrpeter, 2022-09-28:
Hi All
    1. I made a funny thing https://youtu.be/cPjEUe2uY8I
    2. It is a simple verilog trace to show which lines of verilog code is being execute in each clock rise. I want make it better but not sure how to do it:
    3. Are we able to set breakpoint in verilog? any mechanism to do that?
    4. i created a verilog preprocessor using antlr, so it use different verilog grammar than verilator. The preprocessor auto insert code to my verilog and call a c function to record which line has execute. Anyway i can make it better? using two different grammar soon or later has problem
thanks
Peter

wsnyder, 2022-09-28:
Looks useful, good project.

2. This sounds very similar to line coverage which Verilator does support.  Checking what coverage points changed where Verilator inserts coverage would give you the same data.  Note many people use styles with continuous assignments, so large portions of the design will activate every cycle and not tell you anything useful for debug.

3. There is no current breakpointing in Verilog, though I know some commercial simulators support it.  FWIW I don't personally know anyone using breakpoint style debugging on a regular basis because real corportate models are too slow to make it productive.  So this might not get as much use as one might think coming from a software perspective.

4. There are many parsers available, it's unlikely there is any benefit in writing your own.  See e.g. https://github.com/veripool/verilog-perl#which-parser-package  and the scoreboard at https://chipsalliance.github.io/sv-tests-results/

wsnyder, 2022-09-28:
I'm closing as assuming this was mainly informational, feel free to follow up here and if there's something Verilator needs to do specifically please file another ticket.

closed 2022-09-28 by wsnyder