verilator/verilator #3649: Improve warnings to be consistent with gcc

topatopa, 2022-09-30:
This is taken from [this comment](https://github.com/verilator/verilator/pull/3607#issuecomment-1263041809).

I recommend have warnings consistent with [gcc](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html)

> Some options, such as -Wall and -Wextra, turn on other options, such as -Wunused, which may turn on further options, such as -Wunused-value. The combined effect of positive and negative forms is that more specific options have priority over less specific ones, independently of their position in the command-line. For options of the same specificity, the last one takes effect. Options enabled or disabled via pragmas (see [Diagnostic Pragmas](https://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Pragmas.html#Diagnostic-Pragmas)) take effect as if they appeared at the end of the command-line.

The way it is currently implemented in the command line is the last warning appearing in the command line has highest priority. My suggestion:

I think the warnings should be implemented such that each warning has a parent warning. Then the code should traverse a warning until it reaches a root warning and it should prioritize inlined->.vlt->command line->default. This way we can have the warning flags in any order and they will function the same.

wsnyder, 2022-10-16:
Forgot to add earlier... @topatopa would you consider making a pull request to make the changes you recommend?

topatopa, 2022-10-17:
I highly doubt I'll have time to work on this. To be honest, I put it there as a suggestion.

The thing I wanted the most is the splitting of the unused params from unused signals which #3607 achieves.

wsnyder, 2023-04-11:
Closing for now, as not something likely to make progress without a sponsor.

closed 2023-04-11 by wsnyder