verilator/verilator #3655: Support for interfaces as top level ports

kkanhere, 2022-10-04:
Restarting https://github.com/verilator/verilator/issues/1185 discussion here now that interface ports for top level module is supported in linting. I wanted to extend that feature to support interface ports on top level module even for verilation. In order to accomplish this, I think two things need to be supported
1. Control over the parameters of the interface ports - that can be supported via the -G command line feature where the parameters can be uniquely named using the pattern <TOP_MODULE_NAME>_<INTERFACE_PORT_NAME>_<PARAMETER_NAME>
2. Exposing the interface signals as C++ member variables. For this I was thinking the verilated C++ code could generate something akin to the below structure 
```verilog
    interface some_i ();
        logic some_signal1;
        logic some_signal2;
        modport mast_mp (
            input some_signal2
        );
    endinterface

    module some_top (
        input wire some_input,
        some_iface.master_mp some_i
    );
    endmodule
```
converted to
```cpp
    class Vsome_top VL_NOT_FINAL {
    public:
        VL_IN8(&some_input, 0, 0);
        Vsome_i some_i;
       // ...
    };
    
    class Vsome_i VL_NOT_FINAL {
    public:
        VL_INOUT8(&some_signal1, 0, 0);
        VL_IN8(&some_signal2, 0, 0);
    };
```

Essentially the interface can be thought of as another top and instantiated under the top module.

xXIty, 2024-12-10:
Do we know why this was discontinued :((( ?

wsnyder, 2024-12-10:
@xXIty, this issue is still open.