verilator/verilator #3663: VlClassRef should not accept assignment with integers

gezalore, 2022-10-08:
Discovered in #3661 that a variable of type `VlClassRef<T>` can be assigned directly with an integer literal: `VlClassRef<T> foo; foo = 0U;`. This can hide code generation issues so it would be best if this was not possible.

Partial fixes here:
https://github.com/gezalore/verilator/blob/46ed983cf1f29b62095aa95acc8e5dc82bd412e5/src/V3EmitCFunc.cpp#L562-L569
and here:
https://github.com/gezalore/verilator/blob/46ed983cf1f29b62095aa95acc8e5dc82bd412e5/include/verilated_types.h#L1148-L1153

Would @kbieganski be interested in making this proper?

kbieganski, 2022-10-08:
Yes, I'll fix this.

gezalore, 2022-10-08:
A more fundamental issue is that AstConst/V3Number does not have a concept that it represents a nullptr (or any hardcoded address) vs a zero (or a true integer).

kbieganski, 2022-10-08:
What about `V3Number`'s `isNull()`? Not sure how it's used though.

gezalore, 2022-10-08:
Hmm, missed that, might just be what we are looking for.

PR #3671 (2022-10-13): Make class ref typing stricter
Prevents the possibility of assigning an integer to a class reference, both at the SystemVerilog and the emitted C++ levels. Fixes #3663.
closed 2022-10-13 by kbieganski