verilator/verilator #3664: FST waveform dump with Verilator 5.001 and `--binary` option

jeras, 2022-10-08:
```bash
$ verilator --version
Verilator 5.001 devel rev v4.216-37-gca42be98
```
I am trying to get a waveform dump with Verilator 5.001 and `--binary` option, so without a custom C++ source file.
Previously I was specifying the dump file in the C++ `main`.

Since I could not find a CLI approach to specify the dump file, I used the Verilog syntax:
```Verilog
initial
begin
  $dumpvars;
  $dumpfile("test.fst");
end
```

This is the build log with an error:
```bash
verilator --binary -O3 -j 0 --timescale 1ps/1ps --x-assign fast -Wno-INITIALDLY --assert --trace-fst --trace-structs -DTRACE_DEBUG  --top tcb_tb ../../hdl/rtl/tcb_if.sv ../../hdl/rtl/tcb_pas.sv ../../hdl/rtl/tcb_arb.sv ../../hdl/rtl/tcb_dec.sv ../../hdl/rtl/tcb_reg.sv ../../hdl/rtl/tcb_err.sv ../../hdl/rtl/gpio/tcb_gpio.sv ../../hdl/rtl/uart/tcb_uart_ser.sv ../../hdl/rtl/uart/tcb_uart_des.sv ../../hdl/rtl/uart/tcb_uart_fifo.sv ../../hdl/rtl/uart/tcb_uart.sv ../../hdl/tbn/tcb_pkg.sv ../../hdl/tbn/tcb_mon.sv ../../hdl/tbn/tcb_sub.sv ../../hdl/tbn/tcb_man.sv ../../hdl/tbn/tcb_tb.sv ../../hdl/tbn/gpio/tcb_gpio_tb.sv ../../hdl/tbn/uart/uart_model.sv ../../hdl/tbn/uart/uart_model_tb.sv ../../hdl/tbn/uart/tcb_uart_tb.sv
make[1]: Entering directory '/home/ijeras/VLSI/rp32/submodules/tcb/sim/verilator/obj_dir'
ccache g++  -I.  -MMD -I/home/ijeras/VLSI/rp32/submodules/verilator/include -I/home/ijeras/VLSI/rp32/submodules/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++14 -Os -c -o verilated.o /home/ijeras/VLSI/rp32/submodules/verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/home/ijeras/VLSI/rp32/submodules/verilator/include -I/home/ijeras/VLSI/rp32/submodules/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++14 -Os -c -o verilated_fst_c.o /home/ijeras/VLSI/rp32/submodules/verilator/include/verilated_fst_c.cpp
ccache g++  -I.  -MMD -I/home/ijeras/VLSI/rp32/submodules/verilator/include -I/home/ijeras/VLSI/rp32/submodules/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++14 -Os -c -o verilated_timing.o /home/ijeras/VLSI/rp32/submodules/verilator/include/verilated_timing.cpp
/usr/bin/perl /home/ijeras/VLSI/rp32/submodules/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtcb_tb.cpp Vtcb_tb___024root__DepSet_hb637e3ac__0.cpp Vtcb_tb___024root__DepSet_h58356342__0.cpp Vtcb_tb_tcb_if__A20_D20__DepSet_h68ff4973__0.cpp Vtcb_tb_tcb_if__A20_D20__DepSet_h026dc865__0.cpp Vtcb_tb__main.cpp Vtcb_tb__Trace__0.cpp Vtcb_tb___024root__Slow.cpp Vtcb_tb___024root__DepSet_hb637e3ac__0__Slow.cpp Vtcb_tb___024root__DepSet_h58356342__0__Slow.cpp Vtcb_tb_tcb_if__A20_D20__Slow.cpp Vtcb_tb_tcb_if__A20_D20__DepSet_h026dc865__0__Slow.cpp Vtcb_tb__Syms.cpp Vtcb_tb__Trace__0__Slow.cpp > Vtcb_tb__ALL.cpp
ccache g++  -I.  -MMD -I/home/ijeras/VLSI/rp32/submodules/verilator/include -I/home/ijeras/VLSI/rp32/submodules/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++14 -Os -c -o Vtcb_tb__ALL.o Vtcb_tb__ALL.cpp
echo "" > Vtcb_tb__ALL.verilator_deplist.tmp
Archive ar -rcs Vtcb_tb__ALL.a Vtcb_tb__ALL.o
g++    verilated.o verilated_fst_c.o verilated_timing.o Vtcb_tb__ALL.a   -lz    -o Vtcb_tb
rm Vtcb_tb__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/ijeras/VLSI/rp32/submodules/tcb/sim/verilator/obj_dir'
obj_dir/Vtcb_tb 
%Warning: $dumpvar ignored as not proceeded by $dumpfile
%Error: Vtcb_tb.cpp:114: Turning on wave traces requires Verilated::traceEverOn(true) call before time 0.
Aborting...
make: *** [Makefile:129: sim] Aborted (core dumped)
```

wsnyder, 2022-10-09:
Hmm, I thought we had fixed the main() generator for when using --trace. Might you submit a pull to fix it?

jeras, 2022-10-09:
No, sorry. My C++ and Verilator internals knowledge is only up to writing unit tests and maybe trivial fixes.

gezalore, 2022-10-09:
> %Warning: $dumpvar ignored as not proceeded by $dumpfile

Shouldn't this say preceded?

closed 2022-10-09 by wsnyder