verilator/verilator #3666: Segmentation fault with Verilator

flaviens, 2022-10-10:
Hi there!

I encountered a segmentation fault with `Verilator 4.228 2022-10-01 rev v4.228-28-g04fff7ebd` on an Ubuntu 22.04 machine.
[Here is a repository for you to reproduce the issue](https://github.com/flaviens/verisegf). 

Thank you very much!
Flavien

gezalore, 2022-10-10:
Can you confirm if this ever worked on an earlier version? I am getting an internal error on the same all the way back to 4.220

flaviens, 2022-10-10:
Ah sorry the message was not super clearly written in this regard :)
I have only tried with Verilator 4.228 and do not know whether it would work with any earlier version.

Thanks!

gezalore, 2022-10-10:
This is caused by a redefinition of a package:

```
% grep -n "package ariane_soc" src/cva6_pickled.sv
3651:package ariane_soc;
57980:package ariane_soc;
```

These are actually reported at the vert beginning of the output:
```
%Warning-MODDUP: ../src/verisegf_0.1/src/cva6_pickled.sv:57980:9: Duplicate declaration of module: 'ariane_soc'
57980 | package ariane_soc;
      |         ^~~~~~~~~~
                 ../src/verisegf_0.1/src/cva6_pickled.sv:3651:9: ... Location of original declaration
 3651 | package ariane_soc;
      |         ^~~~~~~~~~
                 ... For warning description see https://verilator.org/warn/MODDUP?v=5.001
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
%Warning-MODDUP: ../src/verisegf_0.1/src/cva6_pickled.sv:58080:9: Duplicate declaration of module: 'ariane_axi_soc'
58080 | package ariane_axi_soc;
      |         ^~~~~~~~~~~~~~
                 ../src/verisegf_0.1/src/cva6_pickled.sv:3751:9: ... Location of original declaration
 3751 | package ariane_axi_soc;
      |         ^~~~~~~~~~~~~~
```

While this is invalid input, we should not crash so that needs fixing.

The crash is due to some names being resolved to definitions in the first package, then that package being deleted due to the re-definition, then at a a later pass we try to look for the deleted package as some names were resolved against it.

@wsnyder what do you think is most appropriate here? Section '3.13 Name spaces' of IEEE 1800-2017 say this is invalid (both for packages and modules):

```
a) The definitions name space unifies all the non-nested module, primitive, program, and interface identifiers defined outside all other declarations. Once a name is used to define a module, primitive, program, or interface within one compilation unit, the name shall not be used again (in any compilation unit) to declare another non-nested module, primitive, program, or interface outside all other declarations.
b) The package name space unifies all the package identifiers defined among all compilation units. Once a name is used to define a package within one compilation unit, the name shall not be used again to declare another package within any compilation unit.
```

Can we just make MODDUP a hard error and give early when detected?

gezalore, 2022-10-10:
An alternative is to not delete the original definition, but I think the name clashes might cause problems in later link stages too.

gezalore, 2022-10-10:
Probably the same issue would manifest with this, though I have not tried:

```systemverilog
module sub;
endmodule

module mid_a;
  sub i_sub;
endmodule

module sub;
endmodule

module mid_b;
  sub i_sub;
endmodule

module top;
  mid_a a;
  mid_b b;
endmodule
```

wsnyder, 2022-10-14:
When first bringing up a design MIDDUP does occur regularly so I wouldn't just exit on the first one.

Perhaps on MODDUP we rename one of the duplicates to something else (i.e. add "__MODDUP<incrementednumber>"

gezalore, 2022-10-14:
Are you expecting functional output?

wsnyder, 2022-10-14:
I just expect the normal exit on error to occur (as opposed to immediate fatal).

gezalore, 2022-10-14:
That's what I was suggesting, though possibly at the end of the pass (I do t recall where exactly the exit on error is checked, but currently we go far further than is safe with this type of invalid inpu), will look into it.

wsnyder, 2022-10-14:
Oh, that's ok, I thought you were suggesting a fatal-equivalent.

wsnyder, 2022-10-16:
Duplicate of #3231

closed 2022-10-16 by wsnyder