verilator/verilator #3668: Threads

MartinSolk, 2022-10-11:
Hi,
Do verilator support multthreading, because in the makefile the number which is passed through threads (VM_THREADS) is never used again, except for the #ifneq. Where is the number of threads used?

wsnyder, 2022-10-11:
The number of threads is hard-coded into the Verilated code, see docs/internals.rst.  The ifdef just needs to tell the library if any threads are needed.

closed 2022-10-11 by wsnyder

kbieganski, 2022-10-17:
Also see [the manual](https://verilator.org/guide/latest/verilating.html#multithreading).

MartinSolk, 2022-10-24:
@wsnyder Thanks for your answer, but what i don't understand, if the number of threads is hard coded, why we have the possibility to choose the number of threads.

MartinSolk, 2022-10-24:
Ok, that make sense. But where in the makefile {N} is used if you want to run the threads parallel. I find nothing in the makefile only the if condition for threadsafe or not, as you described

gezalore, 2022-10-24:
Not sure what use you are expecting it's a boolean so rightly it's only used in conditionals

MartinSolk, 2022-10-24:
In the end i want to find out if/how multithreading is possible even when i didn't use the makefile of verilator to build the program. Therefore it would be interesting, where the number of e.g. "--threads 2" is used. :D

gezalore, 2022-10-24:
As Wilson mentioned, the actual number is only used by Verilator internally, and is built into the generated model. If you want to know at run-time how many threads the model uses see the overload of `VerilatedModel::threads()` in the model generated by Verilator (this always returns a number >= 1). When you pass `--thread N`, with N >= 1, the generated makefile also defines VM_THREADS which takes care of the rest when using the standard makefiles. If you want to build with your own makefile, you need to build the model and the run-time library (stuff that resides in `$VERILATOR_ROOT/include`) with `-DVL_THREADED=1` passed to the C++ compiler. The actual threads are created in the VerialtedContext, via VlThreadPool, and is all done automatically if VL_THREADED is defined. Also please read this section of the manual as suggested above: https://verilator.org/guide/latest/verilating.html#multithreading

Note: In the upcoming v5 release, everything will behave as if VL_THREADED is always defined.

MartinSolk, 2022-10-24:
Ok thank you for the explanation, i doesn't know that is done automatically.