verilator/verilator #3669: Verilated objects fails to build when using Macos Monterey

tianrui-wei, 2022-10-12:
Thanks for taking the time to report this.

# Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

The code is at https://github.com/tianrui-wei/verilator-reproduce

Both examples are from using verilator --timing

## when using the system bundled clang compiler

`verilator --timing --cc --exe half_adder_tb.v half_adder.v`
` make -C obj_dir -f Vhalf_adder_tb.mk CC=clang CXX='clang++ -std=c++20 -fcoroutines-ts'`
Gives me the error

```
In file included from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/local/share/verilator/include/verilated_timing.h:52:12: fatal error: 'coroutine' file not found
#  include <coroutine>
           ^~~~~~~~~~~
1 error generated.
```

## When using the homebrew llvm version and the bundled libc++ library

`verilator --timing --cc --exe half_adder_tb.v half_adder.v`
`make -C obj_dir -f Vhalf_adder_tb.mk CC=clang CXX='clang++ -std=c++20`

Gives me the error

```
In file included from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/local/share/verilator/include/verilated_timing.h:98:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:106:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlFileLineDebug fileline)
                      ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:183:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:184:23: error: no matching member function for call to 'push_back'
                queue.push_back({delay, VlCoroutineHandle{coro, fileline}});
                ~~~~~~^~~~~~~~~
/opt/homebrew/opt/llvm/bin/../include/c++/v1/vector:582:66: note: candidate function not viable: cannot convert initializer list argument to 'const std::vector<VlDelayScheduler::VlDelayedCoroutine>::value_type' (aka 'const VlDelayScheduler::VlDelayedCoroutine')
    _LIBCPP_CONSTEXPR_AFTER_CXX17 _LIBCPP_INLINE_VISIBILITY void push_back(const_reference __x);
                                                                 ^
/opt/homebrew/opt/llvm/bin/../include/c++/v1/vector:584:66: note: candidate function not viable: cannot convert initializer list argument to 'std::vector<VlDelayScheduler::VlDelayedCoroutine>::value_type' (aka 'VlDelayScheduler::VlDelayedCoroutine')
    _LIBCPP_CONSTEXPR_AFTER_CXX17 _LIBCPP_INLINE_VISIBILITY void push_back(value_type&& __x);
                                                                 ^
In file included from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/local/share/verilator/include/verilated_timing.h:231:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:246:29: error: no template named 'coroutine_handle' in namespace 'std'
    bool await_suspend(std::coroutine_handle<>) const { return false; }  // Resume immediately
                       ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:256:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) const { coro.destroy(); }
                       ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:290:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) { join->m_susp = {coro, fileline}; }
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:305:14: error: no template named 'coroutine_handle' in namespace 'std'
        std::coroutine_handle<> m_continuation;  // Coroutine to resume after this one finishes
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:313:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never initial_suspend() const { return {}; }
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:317:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never final_suspend() noexcept;
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:350:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) { m_promisep->m_continuation = coro; }
                       ~~~~~^
/usr/local/share/verilator/include/verilated_timing.cpp:161:6: error: no type named 'suspend_never' in namespace 'std'
std::suspend_never VlCoroutine::VlPromise::final_suspend() noexcept {
~~~~~^
13 errors generated.
```

# What 'verilator --version' are you using?  Did you try it with the git master version?

- Verilator 5.001 devel rev v4.228-148-g8dacbdec3
- This is the git master version

# What OS and distribution are you using?

M1 Pro Macbook Pro, Macos Monterey 12.6

# May we assist you in trying to fix this in Verilator yourself?

Yes that'd be great

wsnyder, 2022-10-13:
You selected this compiler

CXX='clang++ -std=c++20 -fcoroutines-ts'

however it does not seem to really support C++20's coroutine_handle as it should.   All I can suggest is use a different compiler, e.g. gcc, which might mean you do not get timing support.

Closing as answered feel free to followup though.

kbieganski, 2022-10-13:
Reopening this for now and assigning to myself. I think I'll need to fix support for libc++ (or get rid of it altogether).

@tianrui-wei Can you try this patch? https://github.com/antmicro/verilator-1/commit/cbee1990636c1ba573fd6ecf613c8e65db332307

I'll open a PR that addresses this but I need some time for testing various configurations.

Some fun facts:
* Apple ships Clang 14 with libc++ 13
* If you use coroutine types from the `std::experimental` namespace, Clang > 13 warns that support for this will be removed in LLVM 15 (though it's still there on LLVM 16 snapshot).
* If you use coroutine types from the `std` namespace, which are available in libc++ > 13, the compiler doesn't find them. That's because apparently [they require C++20](https://github.com/llvm/llvm-project/blob/329fda39c507e8740978d10458451dcdb21563be/libcxx/include/__coroutine/coroutine_handle.h#L23), and the `-fcoroutines-ts` flag is not enough anymore.

closed 2022-10-20 by wsnyder

tianrui-wei, 2022-10-22:
Hi @kbieganski, thanks for the commit! The issue has indeed been fixed :)