verilator/verilator #3676: Regression on v5 master

dpetrisko, 2022-10-15:
Hello, I have encountered a regression moving from v4->v5 of Verilator. I encountered this as part of the BlackParrot compilation, but narrowed down the issue to one specific BaseJump STL module.

>Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
module bsg_encode_one_hot #(parameter width_p=8)
(input [width_p-1:0] i
 ,output [$clog2(width_p)-1:0] addr_o
);

  localparam levels_lp = $clog2(width_p);
  // adjust for non-power of two input
  localparam aligned_width_lp = 1 << $clog2(width_p);

  genvar level;
  genvar segment;

  wire [levels_lp:0][aligned_width_lp-1:0] addr;
  wire [levels_lp:0][aligned_width_lp-1:0] v;

  // base case, also handle padding for non-power of two inputs
  assign v   [0] = i << (aligned_width_lp - width_p);
  assign addr[0] = '0;

  for (level = 1; level < levels_lp+1; level=level+1)
    begin : rof
      localparam segments_lp = 2**(levels_lp-level);
      localparam segment_slot_lp = aligned_width_lp/segments_lp;
      localparam segment_width_lp = level; // how many bits are needed at each level

      for (segment = 0; segment < segments_lp; segment=segment+1)
        begin : rof1
          wire [1:0] vs = {
                           v[level-1][segment*segment_slot_lp+(segment_slot_lp >> 1)]
                           , v[level-1][segment*segment_slot_lp]
                          };

          assign v[level][segment*segment_slot_lp] = | vs;

          if (level == 1)
            assign addr[level][(segment*segment_slot_lp)+:segment_width_lp] = { vs[0] };
          else
            begin : fi
              assign addr[level][(segment*segment_slot_lp)+:segment_width_lp]
              = { vs[0]
                 , addr[level-1][segment*segment_slot_lp+:segment_width_lp-1]
                 | addr[level-1][segment*segment_slot_lp+(segment_slot_lp >> 1)+:segment_width_lp-1]
                };
            end
        end
    end

  assign addr_o = addr[levels_lp][$clog2(width_p)-1:0];

endmodule
```

>What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.001 devel rev v4.224-107-g5869fdf7f (git master)
The previous working version was Verilator 4.221 devel rev v4.220-4-g14d24213.

>What OS and distribution are you using?

Centos 7

>May we assist you in trying to fix this in Verilator yourself?

Happy to help turn this into a formal test case, but an actual fix would depend on the scope. Let me know if you need any more information. Thanks!

wsnyder, 2022-10-15:
@gezalore 

      V3DfgDfgToAst.cpp:257: Should have removed undriven sources

flaviens, 2022-10-17:
Hi there! I encountered the exact same issue with the exact same module today. Thanks for your help!

gezalore, 2022-10-17:
Whoopsie, we don't have enough circular logic in our test suite.

commit 840e26b6 (2022-10-17): Fix incorrect return in DFG decomposition

Fixes #3676
closed 2022-10-17 by gezalore